Clooney82/MavLink_FrSkySPort #3: Doesn't compile if #define USE_SINGLE_CELL_MONITOR set

fnoopdogg, 2015-09-02:


fnoopdogg, 2015-09-02:
MavLink_FrSkySPort.ino: In function 'void setup()':
MavLink_FrSkySPort:391: error: 'Mavlink_setup' was not declared in this scope
MavLink_FrSkySPort.ino: In function 'void loop()':
MavLink_FrSkySPort:405: error: '_MavLink_receive' was not declared in this scope
MavLink_FrSkySPort:407: error: 'Mavlink_check_connection' was not declared in this scope
MavLink_FrSkySPort:417: error: 'FrSkySPort_Init' was not declared in this scope
MavLink_FrSkySPort:425: error: 'FrSkySPort_Process' was not declared in this scope

etc.

Clooney82, 2015-09-03:
Could you please explain more.
For me it was working like it is.
It looks like Mavlink.ino is missing on your local copy.

Znubbis, 2015-09-03:
I have the same problem, if I comment out all addon rows It seems it ignores Mavlink.ino. Compiling with all addons are the only combination I have found working. 

Strange indeed, I'm not used to IDE, but do you not need any "includes" for files in the same directory?

Clooney82, 2015-09-08:
Problem found, but I don´t understand why this is happening.
Problem is if you disabling single_cell_monitor some variable are also removed,
but they are only used by single_cell_monitor.
So I tried a lot of variations, and found out variable LIPOCELL_1TO8 causes the problem.
And it is only used by function LSCM_setup().
If it is also available if you don´t use singe_cell_monitor it compiles without errors.

Clooney82, 2015-09-08:
Fixed with commit: https://github.com/Clooney82/MavLink_FrSkySPort/commit/3be689b10ac460627b9e35671156d4e971a4d682

closed 2015-09-08 by Clooney82

wolkstein, 2015-09-08:
hi clooney,
i am currently working to separate LSCM to its own class. all single cell vars are now inside the class. so maybe the problem is also gone here. i think it would be a good idea when you do not change LSCM things in main branch. then if im done here it is more easy to merge.
the new class will get some more stuff. also to save dividers to eeprom. i will also write some functions to init dividers individual and also an calibration function which auto calculate the right deviders.
see https://github.com/Clooney82/MavLink_FrSkySPort/tree/Refactoring-LSCM-To-own-Class . there is the beginning of LSCM class.
/g
wolke

Clooney82, 2015-09-08:
Hi wolke,

great, will not do any further changes on that.
Only fixed some problems, so that it is possible to disable addons.

wolkstein, 2015-09-08:
ok :),
define of MAXCELL will also removed next. this is the last thing which is currently used in main branch. i do not change it in the new branch because it is used for pre processor #if else on some places. imo we can remove this and use simple if else code. i also create an function to return maxcells. currently commented in LSCM class.

Clooney82, 2015-09-08:
great Idea.
Think it is best to reduce overhead as much as possible.
Also the Idea for calibration.

wolkstein, 2015-09-09:
hi,
today i do some work on LSCM branch.
i merge the last commits from s-c-l-v-rc-opentx2.1 and fix all compile problems with different LSCM definitions like Debug and so on.
IMO the first implementation works and is ready to merge into s-c-l-v-rc-opentx2.1 branch.
if you have time you can double check if it works before merging.

/g
wolke

Clooney82, 2015-09-10:
Pull request merged.