Clooney82/MavLink_FrSkySPort #4: Duplicate FastLED library warning/notification

fnoopdogg, 2015-09-02:
In file included from led_control.ino:18:0:
/Users/siliconalchemy/Documents/Arduino/libraries/FastLED/FastLED.h:12:2: warning: #warning FastLED version 3001000 (Not really a warning, just telling you here.) [-Wcpp]
#warning FastLED version 3001000  (Not really a warning, just telling you here.)

Clooney82, 2015-09-02:
No error, only warning.
Comes from FastLED Library.

fnoopdogg, 2015-09-15:
I guess because it overrides the FastLED library included with the arduino/teensy install.  Is the updated library necessary?
Not important I guess, I'm just a bit anal retentive about warnings, I like cleanliness :)

Clooney82, 2015-09-19:
I prefer to include all used libraries

closed 2015-09-20 by Clooney82

Clooney82, 2015-09-27:
should now be solved with final FastLED 3.1 Library from commit: https://github.com/Clooney82/MavLink_FrSkySPort/commit/c9544299f9bd39ea5d277af98026ecea7627b5ab