Clooney82/MavLink_FrSkySPort #5: Provide .zip GCS_MAVLink library

fnoopdogg, 2015-09-02:
FrSkySportTelemetry.zip and FastLED.zip are in zip form but GCS_MAVLink is in directory form.  Should be zip form with instructions in readme for installation.

Clooney82, 2015-09-02:
Should be no problem to provide Library as a zip file.

Clooney82, 2015-09-08:
fixed with commit: be65a0f
For installing library see: https://www.arduino.cc/en/pmwiki.php?n=Guide/Libraries#toc4

fnoopdogg, 2015-09-15:
For consistency, we should remove GCS_MAVLink directory?

fnoopdogg, 2015-09-15:
Confirmed current GCS_MAVLink directory is exactly the same as the zip file:
> diff GCS_MAVLink tmp/GCS_MAVLink
Common subdirectories: GCS_MAVLink/include and tmp/GCS_MAVLink/include
Common subdirectories: GCS_MAVLink/message_definitions and tmp/GCS_MAVLink/message_definitions

fnoopdogg, 2015-09-15:
Removed directory:
 https://github.com/Clooney82/MavLink_FrSkySPort/commit/c9d47b196897d24a0288f10131dd84a9592d5b05

closed 2015-09-15 by fnoopdogg