Clooney82/MavLink_FrSkySPort #6: Is there a way of setting up the taranis telemetry parameters?

fnoopdogg, 2015-09-02:
Having to set them all by hand is a royal PITA.  It should be automated somehow.

Clooney82, 2015-09-02:
For correct configure it is needed.
Problem / Behavier of opentx 2.1.x.

Maybe we can provide an empty Modelfile.

dk7xe, 2015-09-05:
Hi Clooney, i made an modelfile under companion 2.1.1 with the telemetry settings according your pictures. You can retrieve it from https://drive.google.com/folderview?id=0BxM1Ej9W_KZ7flI0cFJsaGZwTDVId2stSkxQT29OTmhpUF9pc0FleUxkUGl2UWRqQzV5NkE&usp=sharing

Clooney82, 2015-09-09:
I also copied my Model to SD-Card.
Model provided with commit: https://github.com/Clooney82/MavLink_FrSkySPort/commit/3d1298c2bc3707ede7b2cebedd3226188c6b3a18 
Filename: Model000-2015-09-09.bin
Directory: MavLink_FrSkySPort/Lua_Telemetry/DisplayApmPosition/MODELS

Maybe you can test it.

Clooney82, 2015-09-14:
Anyone tested the provided modelfile?
or do we have still an issue here?

fnoopdogg, 2015-09-15:
I setup the parameters manually using your pictures, which works OK although a little confusing.  I did this because I have some nonstandard things setup in my model.  However these are things that should probably all be set for a sane apm FC setup so perhaps supplying them in a model file would help new users.  I'll try blanking/resetting my taranis and trying your model file.

Clooney82, 2015-09-19:
Did you already test the model file?

wolkstein, 2015-09-19:
imo, it is also possible to extract sensor data in lua scripts.
for example "lon" and "lat" are available in "GPS" table. so i think at least there is no need to configure sensors manually. we know all data we send and sensor ids. so it must possible to get them in lua. i will try this next for cells Z1-Zx and Cmin. currently we define custom sensors for this. but a look inside this file https://github.com/opentx/opentx/blob/8b8391101779d85c48e2709c560f8ee78f65111e/radio/src/lua_api.cpp offers other choices to set get them in simple lua.

Clooney82, 2015-09-19:
For Cells, I´m not sure, but for min and max Values it should be possible by adding "-min" or "-max".
like 
```
getValue("VFAS-min")
```

wolkstein, 2015-09-19:
think its more like
``` lua

local MyCells = {} --init simple empty table
local MyCellOne = 0
local MyCellTwo = 0
-- and so on

MyCells = getValue("CELLS") -- do not know but at lua_api.cpp line 166 its telemetry item "cells" 
if  MyCells == table then
    if MyCells[1] ~= NIL then
       MyCellOne = MyCells[1]
    end
    -- and so on for each cell
end
```
i am not really sure but for gps this works. similar to  "if (telemetryItem.cells.count == 0)" on line 166 in lua_api.cpp for cells, you will find for gps something like this "telemetryItem.gps....." .

all this need a try. but if it will work, imo there is no need to setup sensors manually any more.
/g
wolke

mamacher, 2015-09-19:
Hi, have a question and don't know if the is the right place to ask, BTW awsome code, i love this functionality on my Quad... I have a external FrSKY FLVSS and FAS connected to my X8R, I see you just added the option to specify if an external FAS is connected. I actually just added that code the other day and now I can use yours, if I leave the "USE_SINGLE_CELL_MONITOR " commented out will that then pick up my external FLVSS or do i need to do something else to remove the FLVSS from your code..

Clooney82, 2015-09-20:
@wolkstein:
 yeah, would be good if that works. this would save a lot of telemetry slots.

@mamacher:
You are right. Should look like this:
```
/*
 * *******************************************************
 * *** Enable Addons:                                  ***
 * *******************************************************
 */
#define USE_FAS_SENSOR_INSTEAD_OF_APM_DATA              // Enable  if you use a FrSky FAS   Sensor.
//#define USE_SINGLE_CELL_MONITOR                         // Disable if you use a FrSky FLVSS Sensor. - Setup in LSCM Tab
//#define USE_AP_VOLTAGE_BATTERY_FROM_SINGLE_CELL_MONITOR // Use this only with enabled USE_SINGLE_CELL_MONITOR
```

Clooney82, 2015-09-24:
I will close this, due to the fact we can't change this.
We can only improve in small parts, like GPS and maybe Cells.

If someone still thinks we need this, please reopen or open a new issue.

closed 2015-09-24 by Clooney82

davidbitton, 2015-11-10:
@Clooney82 Should the Wiki be updated to include the Discover new sensors step?

athertop, 2015-12-30:
I just left a comment on another issue about GPS sensor but having spotted this issue thought it would be good to share my findings and method for easier sensor setup. I started tonight by clearing all sensors on my Taranis (I was testing the inverter native FrSky telem available now from APM 3.3 and cleared them to switch back to the Teensy), and I chose 'Discover new sensors' to add the majority of sensors back on to my model. All I had to edit once done was to rename Temp1 -> T1 and Temp2 ->  T2, then add the calculated sensors (like Cmin). I also discovered tonight that doing the discovery of sensors this way is the only means of getting a working GPS sensor configured. Manually setting up GPS either on the taranis directly, or in Companion then downloading, results in a broken GPS sensor (it displays just a 5-6 character numeric rather than the Lat/Long values as expected.