Clooney82/MavLink_FrSkySPort #10: All telemetry slots full

dk7xe, 2015-09-05:
i have read the conversation on this topic on the diydrones forum and based on the explanation given this is caused by duplicate sensor stuff.

I started with a clean model (model file available on https://drive.google.com/folderview?id=0BxM1Ej9W_KZ7flI0cFJsaGZwTDVId2stSkxQT29OTmhpUF9pc0FleUxkUGl2UWRqQzV5NkE&usp=sharing ) and entered all the sensors manually according the pictures provided by Clooney. My understanding is that the error should not come up in this case, right?

But it does. Environment used is EeePC1201N / Opensuse13.2_64 / companion 2.1.1

Since i don't have my Teensy3.1 programmed yet i purely do the tests with companion simulator.
Error message appears when i start simulating parameters and enter something (other than RSSI or A1).
[image]
[image]

Clooney82, 2015-09-08:
I think this could be a problem of openTX simulator.
Didn´t test Telemetry Simulator till now.

Clooney82, 2015-09-14:
Fixed with openTX 2.1.2:
>Telemetry sensors are no longer continuously detected, it is now necessary to go to the telemetry setup page and manually select the “Discover new sensors” menu item. This prevents creation of non-existing sensors in case of transmission errors, and allows excluding sensors that are received but considered uninteresting.

closed 2015-09-14 by Clooney82