Clooney82/MavLink_FrSkySPort #9: which one is the correct telem1.lua?

dk7xe, 2015-09-05:
under /SCRIPTS/TELEMETRY there is a telem1.lua and telem1_dev.lua
if i copy both into the SDcard structure companion sees only the telem1 because filename telem1_dev is too long.

Please keep the one and only version (that is further developed) within the directory.

For the time beeing i assumed that telem1_dev.lua is the latest one. I have renamed telem1 to telem0 and telem1_dev.lua to  telem1.lua. Will start testing with this one.

Clooney82, 2015-09-08:
telem1.lua is current working version.
telem1_dev is used for complete redesign of code, and currently not for productive use, only for devs.

closed 2015-09-08 by Clooney82

dk7xe, 2015-09-08:
Hi Clooney,
Thanx for clarification.
Wouln't it make more sense to rename the file to something like "devxxx" to stick to the 6 character rule?

BTW, i's like to contribute to dev (after i managed to understand LUA a bit more). Have some experience with VBA and Linux shell scripting..