Clooney82/MavLink_FrSkySPort #11: Create wiki

Clooney82, 2015-09-08:
Wouldn´t it be great to have a wiki for this project?

dk7xe, 2015-09-08:
Indeed, i was thinking about the same while driving home tonight. Because i was struggling with which telemetry value is representing what. I know it's in the code but i would like to start with creating an overview on that. If you like i can take over this part.

Clooney82, 2015-09-08:
Feel free to contribute ;)

Clooney82, 2015-09-09:
I added first wiki entries.

Clooney82, 2015-09-19:
Please update wiki for current development.
Describe your LUA Script, and how to configure them.
Also if something change on teensy side.

dk7xe, 2015-09-25:
i have updated the [1.2.1 Telemetry Value Overview ](https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.2.1-Telemetry-Value-Overview) page with the information based on what i have found on 1.2. FrSky Taranis Telemetry.
The values Cels, GPS, GSpd and SWR needs a review still

Please let me know if this is sufficient and feel free to update ;-)

Gerald

Clooney82, 2015-09-28:
Looks good so far.
Now we must keep everything up-to-date.
Maybe we need to add more detailed instructions to setup everything.

athertop, 2016-04-05:
I would be happy to get involved in updating the Wiki. I can certainly add info on the different lua script versions now if that is OK with you guys?

Clooney82, 2016-04-05:
Feel free to make modifications ;) and welcome on bord.

athertop, 2016-05-30:
@Clooney82 - I have done quite a bit of work on the Wiki - on my own fork initially before copying it over here. I'm not quite finished with it yet, but before I copy it, I would really appreciate if you have the time, to check out the new Wiki content - kind of proof read it from a technical perspective if you will?

(you will have noticed that I added extra images to the images folder on this repository. I put them here purposely - as I have used these in my updated pages, and wanted to make sure that when I migrate the wiki over here, that all the references carry over OK)

My wiki/fork is here: https://github.com/athertop/MavLink_FrSkySPort/wiki

I have re-written/amended all the pages up to (but not including) 2.1.2 - I still have the final few pages to work on. I won't be touching the LED page at all, as I have little knowledge of this area.

Clooney82, 2016-05-30:
@athertop - looks good. for my good to understand. think now some parts become more clear.

athertop, 2016-05-30:
I still have a fair bit of work to do on it, so will let you know when 
I'm near completion. Thanks for checking it.

athertop, 2016-08-22:
OK, I have merged my changes into the Wiki. Please feel free to check and comment. Thanks, Paul

closed 2016-09-11 by athertop