Clooney82/MavLink_FrSkySPort #15: Phenomenally irritating claxon on by default

fnoopdogg, 2015-09-13:
I've disabled by removing SOUNDS/en/ALARM* files.  I remember other people have also reported the same thing, I assume it's something that's on by default for the single cell code?

wolkstein, 2015-09-13:
they used on two scripts. first is cellinfo. and secomd telm1. in telem1 wh gauge need the info of max wh for example 130. telem1 lua play an ALARM sound if the max value is reached. if i am right telm1 init maxconsume as 0. in code it ask for globle var to set maxconsume. and telm1 play a sound, exactly SOUNDS/en/ALARM3K.wav if calculated watt hours bigger than  maxconsume. so if you do not set all values in offset lua maxconsume is init as 0. this means the alarm ALARM3K.wav play immediately after starting taranis. but whgauge is not a  single cell addon. it also works with normal mavlink reported battery infos. of course it need voltage and current sensor.
/g
wolke

fnoopdogg, 2015-09-15:
changed default to no klaxon unless maxconsume (or whconsume) set in telem1.lua script:
https://github.com/Clooney82/MavLink_FrSkySPort/commit/97f337b07cfbb670e807d09299d0da52854125cc

closed 2015-09-15 by fnoopdogg