Clooney82/MavLink_FrSkySPort #16: Milestones

Clooney82, 2015-09-14:
I´d like to implement Milestones for major changes on the source.
What do you think about this idea?
Maybe you can also bring your ideas for milestones in.

wolkstein, 2015-09-14:
this is a good idea. first milestone can be "stable lua telemetry for >= opentx 2.1.3. this is released sinc a few days, and seems to be change problems on the telemetry sensor detection side.

imo we are fine if we support opentx >= 2.1.3 and leave 2.1.1.

/g
wolke

Clooney82, 2015-09-14:
I´m running AC3.3-rc11 and opentx 2.1.3.
For me it looks pretty good, only smaller issues.

Will add the following milestones:
* Version 1.6 (stabilize)
* Version 1.6.1 (fixes)
* ...
* Version 1.7 (LUA redesign)
* Version 1.7.x (fixes)

Any other ideas ?

wolkstein, 2015-09-14:
ok fine,
but i only jump to AC3.3 if it is really stable. because i use pixhawk on my commercial used copter for aerials.

some questions.
sensor rpm is used for severity messages. debug output on teensy give me rpm values, also my taranis telemetry tab show me an rpm sensor. did the current used lua script located under  https://github.com/Clooney82/MavLink_FrSkySPort/blob/s-c-l-v-rc-opentx2.1/Lua_Telemetry/DisplayApmPosition/SCRIPTS/TELEMETRY/telem1.lua
play the right audiofiles?

my taranis is silent and only play the flight mode wav files.

second,
heading will only send if gps status is 3. this mean no heading infos will send if there is no gps 3d lock. i do not like this so much. imo compas heading which is used here, can be send without 3d lock on gps.

/g
wolke

wolkstein, 2015-09-14:
ok,
got it. forgot to configure Z1 to Z6 and Cmin. RPM was also wrong here. was not RAW-.
oops :).

Clooney82, 2015-09-19:
For implementing new features, please use branch: [s-c-l-v-rc-opentx2.1-next](https://github.com/Clooney82/MavLink_FrSkySPort/tree/s-c-l-v-rc-opentx2.1-next)

fnoopdogg, 2015-09-20:
I'd like to suggest a milestone of moving to official diydrones github home.  This is a small but significant project that contributes an important function to APM.  Craig Elder has already agreed to it, and I think it would be a great milestone to this project, what do you guys think?

Clooney82, 2015-09-20:
like I said on diydrones, I think this is a good idea and will give us more attention.
But I think all of us have to support this.

closed 2015-10-07 by Clooney82