Clooney82/MavLink_FrSkySPort #17: some Telem issues after first test

wolkstein, 2015-09-15:
hi,
today i test the telem on opentx2.1.3.

issues that i noticed on telm1 screen.

*  flight timer do not toggle from armed status. - armed status? - severity messages?
* no  severity messages. - RPM sensor?
* no gps satellite cont.
* heading is only refreshed if gps status is 3

/g
wolke

Clooney82, 2015-09-15:
For me all works.
only severity Messages are untested on my side.
Did you check debug messages?

Btw. I'm this week on a training, so I can't support very good.

lvale, 2015-09-15:
Messages have been redone on 3.3 and still not "closed" yet. Latest messages were on a commit for Simple and SuperSimple mode.

As an extra suggestion. we must drop the trick of having APM values inside not related FrSky messages, like it+s done for RPM, and create new virtual sensors. I had a conversation with Bertrand some months ago and they were looking at having a list of relevant data from APM to reserve SensorID's on FrSky telemetry 2.1.x

Clooney82, 2015-09-15:
Should be no Problem to create new sensors.
I thought already about it, but first i would have a working version, before do bigger changes. But pls Open a new issue for this.

wolkstein, 2015-09-15:
so,
fixed:)
*severity messages
i correct some problems with  severity messages. this was caused by our new FrSkySportSensor library. RPM T2 only refresh all 1000 ms. this was to slow for old implementation. now messages are queued and will send a time of 1400ms. that ensure that RPM T2 handle all  queued messages correctly.
in lua telem1 i correct the file path. now all currently available messages are played. apm3.2 messages. new apm 3.3 messages for example ekf warnings must added.

*gps count was not broken. it was caused by local misconfiguration of telemetry sensors.

*heading need some love.
i do not change the heading issue currently.

at least i have an working telem1 screen.

to-do,
fix cellinfo.lua to play battery warnings and alarm correct.

/g
wolke

closed 2015-09-18 by wolkstein