Clooney82/MavLink_FrSkySPort #18: Duplicate telem1.lua scripts

fnoopdogg, 2015-09-15:
./Lua_Telemetry/DisplayApmPosition/SCRIPTS/Model_Name/telem1.lua
./Lua_Telemetry/DisplayApmPosition/SCRIPTS/TELEMETRY/telem1.lua

Do we still keep the Model_Name directory/scripts around to support opentx-2.0?  I suspect we should just delete it and only support opentx-2.1 going forward?

Clooney82, 2015-09-19:
If it still works on opentx 2.0.x, I would keep it.

wolkstein, 2015-09-19:
yes, we can keep it for 2.0.x users!
it is a really pain to update to 2.1X. i think many users stay at 2.0.x because the latest version 2.0.17 is rock-solid and there is simple no need to update.
/g
wolke

Clooney82, 2015-09-19:
so current version also working with opentx 2.0.17?

wolkstein, 2015-09-20:
good question, i currently finished my update to 2.1:). do not want to switch back for some tests. imo on 2.0.17 i use s-c-l-v-rc branch and not s-c-l-v-rc-opentx2.1

/g
wolke

fnoopdogg, 2015-09-20:
As opentx-2.1 is so different regards telemetry, once it's stable and our companion code is stable for it, do we want/need to continue support for 2.0?  I assumed the s-c-l-v-rc-opentx2.1 was just for 2.1, going forward.  Or maybe a lot of people will continue with 2.0 so we should continue to support it?
At this point perhaps we should rebase master from s-c-l-v-rc-opentx2.1 and drop that branch?

Clooney82, 2015-09-20:
No.

markmach, 2015-09-20:
I keep switching back and forth..  Between 2.0.17 and 2.1.3.. When I find another issue in 2.1 I usually go back to 2.0. I would be fine with just 2.1 branch updates, that would force me to stick with 1 version. :)

Clooney82, 2015-09-20:
From my side, I´ve not planned to update "old" branch, only fix important issues.
But I want to keep it for users who are not will, or can not use new branch.

closed 2015-09-22 by Clooney82