Clooney82/MavLink_FrSkySPort #19: Strip whitespace?

fnoopdogg, 2015-09-15:
Some of the current code (eg. telem1.lua, at least) has quite a lot of trailing whitespace even in empty lines.  Default git editor atom strips superfluous whitespace by default so editing files in atom creates diffs /commits with a lot of changes.  Do we allow this, or do we ignore it (atom for example you can disable the trailing whitespace plugin), or do we use this issue to sweep through the code and remove all superfluous whitespace?

wolkstein, 2015-09-15:
i like the git default whitespace handling. maybe we have to declare some simple code formatting roles. than it will easy to set-up your preferred editor. also we can format the existing code as soon as possible. currently there are only just a less files in this project.
/g
wolke

fnoopdogg, 2015-09-18:
https://github.com/Clooney82/MavLink_FrSkySPort/commit/1e5a1eaf2a03f810d94967c38e20e6ff6dea3afc
How does this look?
I don't pretend by any means to be any expert code and I'm not very familiar with git/opensource development, so please let me know if any of the changes I'm making aren't appropriate.  I don't get offended easily :)

fnoopdogg, 2015-09-18:
Maybe using the apm style/contribution guidelines might be a good idea?
http://dev.ardupilot.com/wiki/style-guide/
http://dev.ardupilot.com/wiki/submitting-patches-back-to-master/

wolkstein, 2015-09-19:
lower cases everywhere? ok, but i am not a friend of them.
underscores in functions make reading with pointers to a pain. 
example:
my_class->my_class_function_to_do_something()
my_class.my_class_function_to_do_something()
vs:
MyClass->MyClassFunctionToDoSomething()
MyClass.MyClassFunctionToDoSomething()

for c++ i like this style guide
https://google-styleguide.googlecode.com/svn/trunk/cppguide.html
/g
wolke