Clooney82/MavLink_FrSkySPort #21: Hi,

wolkstein, 2015-09-16:
i locally use in my telem1 script timer2 in manual reset mode to record over all motors armed time.
i do this because after 80 hours of motor runtime i change all bearings on them.
maybe this feature is only interesting if people fly in a commercial way and have to setup correct pre-post flight checklists also beside to follow maintenance intervals by default.

but i like this feature this really help to remember to check essential mechanical parts.
i use it no to remember me at 80 hours to do an full inspection and motor service.
than the timer reset to zero.
all 5 hours of flighttime it remember me to check all screws.

of course it is possible to increse services remember actions.
sice 2.1.2 timer go up to 157 hours. befor it was limited to -8 - + 8 (16) hours.

if people like this i can add it to our repo.
telem1 screen shot:
[image]

/g
wolke

Clooney82, 2015-09-16:
From my side, why not.

Other question: did you test telem1_dev.lua?

wolkstein, 2015-09-16:
not yet,
i use the last days to port my copter to opentx2.1 and also to upgrade the teensy on this system. than i need around two days to understand 2.1 telemetry setup. now i am happy that it run stable here.
till next week friday i have to mange 4 film and foto jobs, also the system must fly stable on three of this jobs. one is at Möhnetalsperre:). i think after this, i can contribute to the new lua script.

/g
wolke

wolkstein, 2015-09-18:
added in commit 430f6023b1b0cd0e1a5b65498b145ae49300ea09 .

closed 2015-09-18 by wolkstein