Clooney82/MavLink_FrSkySPort #23: to much severity messages.. or to slow?

wolkstein, 2015-09-17:
severity messages currently flute taranis. after my changes on teensycode, commit: 057c3a6dd00b2507ddec8202f49454c497a7df86 , all severity messages will send to taranis. at least this is very good, but the cons, this are a lot messages. after arming you hear around five messages.
calibration- beside init- and ready- messages. after this flightmode and at least the armed message. this need arround 10-15 seconds. at least the vehicle is armed while taranis is still reading pre arm messages. in my case the motors spin since 7 seconds befor taranis is ready with all pre arm messages an say arming. this is not really a problem but a liite bit a feeling like an audio offset in films. you see something but you hear it later.    

currently teensy repeat each message 1400ms. this produce at least a small time gap on taranis between messages. this mean reducing this to 1100ms save some time without the risk to loose messages because RPM T2 only transmit each 1000ms. but this recover only 2 seconds.

any other ideas?

/g
wolke

Clooney82, 2015-09-20:
Haven´t tested it until now, but maybe we can play only important messages.

wolkstein, 2015-09-20:
yes, but user configurable from an mixer script. because for example i need infos about waypoints. i fly a lot of jobs where i have to use navigation points. and it is really important for me to get informed if an command member is reached.
so maybe we can create an mixer script which do the severity job with some filter options which can enabled/disabled.

so imo, we can move this issue to milestone 1.7. currently i think we can reduce the dequeue time from 1400ms to 1000ms or 1010ms. it is only important that we send inside the 1000ms interval for T2. that will be enough for milestone 1.6.
/g
wolke

wolkstein, 2015-09-26:
testing with 1050 up to 1300 ms lost randomly some messages. at 1400ms all messages are captured.

at 1400ms we also have randomly messages twice. this need a bit more debugging. i am not sure if this happens because FC send different messages but our parser failed. i use rev apm3.2, so imo i am on the old message system.

because messages are queued we can simple remove double messages from queue. but before we do such filtering on teensy we need more infos if the parser works as expected.

/g
wolke

Clooney82, 2015-09-26:
@wolkstein maybe you can test my latest commit https://github.com/Clooney82/MavLink_FrSkySPort/commit/41e6a5aba6b079bcc7e68b95865ed1df0e25eeb7 in my lua_rework branch.
I added possibility to send severity messages to RC

wolkstein, 2015-09-28:
because i use 6s all my telemetry slots are in use on taranis. first i have the need to figure out how i can get cells 1-x without self defined custom sensors on taranis telemetry.