Clooney82/MavLink_FrSkySPort #24: Disable battery voltage, current and consumption

Luftpumpa252, 2015-09-17:
Hello, please i need disable battery voltage, current and consumption. I use Frsky Fas 150 and FLVSS sensors and its colide with pixhawk data.

Clooney82, 2015-09-19:
fixed with commit: https://github.com/Clooney82/MavLink_FrSkySPort/commit/8effa34343ccbed23645e66a8e6f87739d72acd5

See also wiki -> https://github.com/Clooney82/MavLink_FrSkySPort/wiki/2.-AddOns

ggregory8, 2016-05-18:
Sorry to bring up this old issue, but I need clarification. If we enable option USE_FAS_SENSOR_INSTEAD_OF_APM_DATA do we still get APM data for the rest of the telemetry?

I.e can we have FLVSS connected as well as APM Mavlink data? I will need to see how FLVSS connects to the SPORT with teensy,  i.e is HUB needed.

athertop, 2016-05-19:
ggregory8,

Yes, you can have FLVSS as well as APM Mavlink data - you do not need to do any code changes from the supplied ones in the current code for this to work. The sensor created by FLVSS on the Taranis will be called 'Cels'. This can be used to create a calculated sensor in 2.1 called 'Cmin' which displays the minimum Cell voltage reading in the LUA telemetry page. With your mention of a HUB, I am not sure if you are getting confused about the FLVSS sensor - FLVSS is a smart-port sensor (digital) and does not use a HUB, but is simply daisy-chained with other sensors (such as the teensy board) before eventually being connected to an 'X' series FrSky telemetry receiver. The HUB you mention is what used to be used with the analog sensors on the D series receivers, and has no relation to the smart-port sensor system used by the X series receivers and sensors, of which FLVSS is one, and upon which this project is based.
The Cels telemetry sensor generated by the FLVSS is not however used to display the battery voltage on the telemetry page, as this data comes from the 'Vfas' sensor - which is derived from Mavlink (ultimately from the APM Power module if you have one of these connected).
This means that your top-left voltage display will be APM sensor derived and your V C-min reading will be derived from the FLVSS sensor. I prefer it this way as knowing what the Pixhawk voltage reading lets me know how close I am getting to a battery failsafe. Both the Cels sensor and Vfas sensor values should be pretty close though. If you prefer to have the Cels value displayed on the telemetry screen instead of Vfas then you can easily amend the LUA script for that to be used instead, or alternatively, delete the existing Vfas sensor on the Taranis and rename the Cels sensor to Vfas, making sure you amend that calculated Cmin sensor to be now based on the new Vfas sensor.

closed 2016-05-19 by athertop