Clooney82/MavLink_FrSkySPort #25: Fix mavlink setup

fnoopdogg, 2015-09-18:
Longstanding bug where if APM serial parameters are not set to broadcast unconditionally then we don't request/connect.  Currently on the pixhawk as of apm:copter-3.3, Telemetry1 and Serial4 are set to broadcast unconditionally where as Telemetry2 isn't (who knows why), which causes confusion.
Users shouldn't have to set SR2_x (or SR1_/SR4) in APM firmware parameters, it should be requested by us.

Clooney82, 2015-09-20:
Maybe you can open an issue on ardupilot side.
Normally it should work without setting these [parameters](https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.1.-Teensy-and-APM-pixhawk), as we tell the FC to send mavlink data we need.
See [mavlink.ino](https://github.com/Clooney82/MavLink_FrSkySPort/blob/s-c-l-v-rc-opentx2.1/MavLink_FrSkySPort/Mavlink.ino) lines 60 - 90.

Clooney82, 2015-09-24:
any updates here?

fnoopdogg, 2015-09-24:
I'll try and have a look today.  I spent a while looking at this before and refactored some of the code, but couldn't obviously see why it wasn't working.  Will try again.

Clooney82, 2015-09-24:
I think problem is not on our side, think it´s AC related.
Because we tell AC what we want, and it does not work always.
I also invested some time in this part when redesigning code, but I could find a solution for this.

fnoopdogg, 2015-09-24:
OK well it's working now as of 3.3-rc12.  I reset all APM parameters and verified SR2_ parameters all set to 0 by default.  Fired up pixhawk (and hence xr8/teensy), refreshed parameters and verified that SR2_EXT_STAT, SR2_EXTRA2, SR2_RAW_SENS are set to 10, as set by teensy.  Reset SR2 parameters back to 0 and tried again, successfully.  Note that Serial4 on pixhawk is set to protocol 5 baud 38 by default - it's set for secondary GPS - but if reset to protocol 1 and baud 57 it works also for teensy.  SR4_xxx parameters don't exist, APM issue raised here:
https://github.com/diydrones/ardupilot/issues/2895

closed 2015-09-24 by fnoopdogg

Clooney82, 2015-09-24:
OK, hope this is really solved. Will try if I find time to test.