Clooney82/MavLink_FrSkySPort #26: remove unused images and sound file

Clooney82, 2015-09-20:
We hava a lot of images and sound files provided. 
Not all are used, maybe we can sort out images and sound file which are not used.

And also, we could regenerate sound files completely, to have a consistent voice.

wolkstein, 2015-09-20:
the sound files i generate was created with http://www.oddcast.com/home/demos/tts/tts_example.php .
for some reasons i create different sounds to ensure that some messages simple get more ore other attention. most files are created by lvale. i think he also have in mind that he use for some messges also different voices filter for listeners.

/g
wolke

fnoopdogg, 2015-09-20:
Funny, I had already started work on this in the background.  I'm creating an automated way of pulling messages and generating soundfiles, using a variety of speech synthesisors and voices.

wolkstein, 2015-09-20:
the open-source text to speech synthesizer i know and also i am using some times, is festival. it is really easy to use in console. also it is easy to write simple scripts. but the results are imo not so nice than the results from online text to speech synths.

which synths did you use?

fnoopdogg, 2015-09-20:
Basically wrappers around the google tts, like:
 https://github.com/desbma/GoogleSpeech
The quality is the best that I found so far.

wolkstein, 2015-09-20:
cool, desbma is licensed under GPLv3. need at least some good sound fonts. do you know if snd files generated with google tts are free to distribute?

lvale, 2015-09-20:
There is a new good option now.
https://play.google.com/store/apps/details?id=com.adrionics.android.opentx.speech

although I still prefer using MacOSX TTS voices.

Clooney82, 2015-09-20:
I create soundfiles  with OS Tools, like here described: http://openrcforums.com/forum/viewtopic.php?f=64&t=6871

This is very easy for me, and generates files in background. Only Input as csv file is needed.

fnoopdogg, 2015-09-20:
A further problem with messages now is that copter-3.3 doesn't just have fixed text strings, it has variable prearm messages generated by EKF.  I think we probably need to keep a lookup table of static messages as before, but also have a separate EKF/Vibe function as per the new GCS functionalities.

Clooney82, 2015-09-21:
For Soundfile generation I need a csv file in the following format:

Path | Filename | Language (EN) | Language (DE) | Language XY
-------|--------------|----------------------|----------------------|--------------------
/system | 0000 | 0 | 0 | 0
/system | rssi_org | RSSI low | RSSI niedrig | ...

For a complete sound pack I need about 5-10min, but my own translation is currently not complete.

Clooney82, 2015-09-28:
Just pushed my soundgen to github: https://github.com/Clooney82/MavLink_FrSkySPort/commit/1eb574cb115413e48e3936ff193bda14a4d3f603

lvale, 2015-09-28:
Good job. I believe that for "normal" users a prebuilt sound pack in English EN-US would be better with all the sounds using the same voice. I tend to use AVA as the voice for all system sounds and also for all the messages.

Clooney82, 2015-09-29:
think for new sound we should use a seperated folder, like 
/SOUNDS/*language*/**TELEM**/
or
/SOUNDS/*language*/**mavlink**/

closed 2015-10-21 by Clooney82