Clooney82/MavLink_FrSkySPort #27: Add a telemetry screen for messages, add mute option

fnoopdogg, 2015-09-20:
It would be nice to have a visual messaging system with the option of muting voice alerts.  There are a few scenarios where voice alerts disturb the environment (either people or animals, and my wife during development).  It would be great to have either a single line on the main telemetry screen for displaying current messages, or else a separate telemetry screen (there used to be one, it was very useful for displaying history of messages on taranis while flying).

mamacher, 2015-09-20:
It would also be nice to have another telemetry screen to display all messages incase you miss the current message on screen

Clooney82, 2015-10-21:
Mute option is now added in s-c-l-v-rc-opentx2.1-next branch.
Displaying Messages is still open. Had an idea how to implement this, but this is not working in openTX 2.1. maybe I´ll find a solution in future.

Clooney82, 2015-10-30:
Now it is possible to show text messages with next branch on Taranis.
See commit: https://github.com/Clooney82/MavLink_FrSkySPort/commit/91f502dddbd6f9c6ed4490bfa33415700bc97cf2
But you need a new Telemetry Sensor on Taranis.
Name: MSG
Type: Sensor
ID: 0600 6
Unit: %

lvale, 2015-11-10:
Just as a heads-up, regarding the messages from the flight controller.
I've just submitted a large commit to ArduPilot master, that uniformizes all the messages sent by the flight controller, to the standard set by MAV_SEVERITY

I believe this makes it quite easier to parse and decide which should be displayed and/or spoken.

Severity Debug (7) is not used

All the PreArm and Arm prefixes are still on Critical severity.

0	MAV_SEVERITY_EMERGENCY	System is unusable. This is a "panic" condition.

1	MAV_SEVERITY_ALERT	Action should be taken immediately. Indicates error in non-critical systems.

2	MAV_SEVERITY_CRITICAL	Action must be taken immediately. Indicates failure in a primary system.

3	MAV_SEVERITY_ERROR	Indicates an error in secondary/redundant systems.

4	MAV_SEVERITY_WARNING	Indicates about a possible future error if this is not resolved within a given timeframe. Example would be a low battery warning.

5	MAV_SEVERITY_NOTICE	An unusual event has occured, though not an error condition. This should be investigated for the root cause.

6	MAV_SEVERITY_INFO	Normal operational messages. Useful for logging. No action is required for these messages.

7	MAV_SEVERITY_DEBUG	Useful non-operational messages that can assist in debugging. These should not occur during normal operation.

Clooney82, 2015-11-10:
could you pls give us a link to your commit?

lvale, 2015-11-10:
Here goes:
https://github.com/diydrones/ardupilot/pull/3118
https://github.com/diydrones/ardupilot/pull/3117
https://github.com/diydrones/ardupilot/pull/3116
https://github.com/diydrones/ardupilot/pull/3112

or

https://github.com/diydrones/ardupilot/pulls?q=is%3Apr+is%3Aclosed+author%3Alvale

athertop, 2016-12-05:
Initial implememtation of a messaging screen just committed. No mute option implemented yet in 2.1 repo.