Clooney82/MavLink_FrSkySPort #29: Add a proper License model to this project.

wolkstein, 2015-09-20:
we need an license model for this project.
imo,
c/c++ and lua code can licensed under gpl3 (GNU GENERAL PUBLIC LICENSE)

and images snd-files and other documents can distributed under cc license. and in this case not free for commercial use.
like this:
http://creativecommons.org/licenses/by-nc/4.0/
[image]

in our case we have to talk with
https://github.com/chsw (Christian) who starts the project 
and
https://github.com/lvale (Luis Vale) who add a lot of snd files and images. also our telm1 script is at least a copy from him. i modified a lot but the base layout are from him. i create the radar and the Wh gauge.

/g
wolke

Clooney82, 2015-09-20:
agree.
Luis can answer directly, but who will ask chsw.
And maybe we have to take care of [S.Port Library](http://www.rcgroups.com/forums/showthread.php?t=2245978#post29439177) of pawelsky.

wolkstein, 2015-09-20:
currently it is your repository, maybe you ask chsw. this will the direct way. but i can also ask him.

fnoopdogg, 2015-09-20:
chsw didn't start the project - he added the original code to github and improved it.  The original code is from:
 http://diydrones.com/profile/RolfBlomgren
His code wasn't released with a license or copyright but was derived from:
 https://code.google.com/p/telemetry-convert/
which is gpl3.  
Luis' lua work is gpl3 as well.

Clooney82, 2015-09-20:
As if Rolf project started with this, we should also us this, because I think we can´t change licensing.
Commerical usage should be not allowed without contacting us.

wolkstein, 2015-09-20:
this mean we can add gpl3 licence to
FrSkySPortTelemetry.ino
MavLink_FrSkySPort.ino
Averange.ino
or not?

new  Mavlink.ino is from Clooney?
also the led stuff.

lvale, 2015-09-20:
Hi

I can ask Christian (chsw) if needed, we’ve spoke a lot when we started the project, then we chose to split it specially due to the different lua scripts that were causing confusion on users (and my inability to deal efficiently with GitHub :)  )

Sometime ago I added a generic GNU license to my repo, because I saw some guys selling pre-programmed teensy’s without attribution, but that can be expected even from this project :( , because even with the license on the script those guys persisted…..but…. life’s too short to worry with that…..

Any licensing that preserves the original code and discards responsibility for using the code is a good license.

Clooney82, 2015-09-21:
S.Port library is Not for commercial use.

wolkstein, 2015-09-22:
@lvale,
what did you think about "cc" for snd-files and images. i think you created most of them.

what all think about this header in, FrSkySPortTelemetry.ino, MavLink_FrSkySPort.ino, Averange.ino.

```
MavLink_FrSkySPort
by
Original Author: Jochen Tuchbreiter (2013)  under (GPL3)
https://code.google.com/p/telemetry-convert/

(2014) Improved by:
   Rolf Blomgren
   Christian (chsw, i do not now the name)
   https://github.com/chsw/MavLink_FrSkySPort

   Lvale
   https://github.com/lvale/MavLink_FrSkySPort

   Michael Wolkstein 
   https://github.com/wolkstein/MavLink_FrSkySPort
   
   and so on till all involved authors are called. imo this is what gpl3 is expect. calling all authors by name if you use the code and republish them.

https://github.com/Clooney82/MavLink_FrSkySPort 

at least we add the gpl3 licenses text like this.

 This program is free software; you can redistribute it and/or modify
 it under the terms of the GNU General Public License as published by
 the Free Software Foundation; either version 3 of the License, or
 (at your option) any later version.
 
 This program is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY, without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
 GNU General Public License for more details.

 You should have received a copy of the GNU General Public License
 along with this program; if not, write to the Free Software
 Foundation, Inc., 59 Temple Place, Suite 330, Boston, MA  02111-1307  USA
  
  ```

Clooney82, 2015-09-22:
looks pretty good.

lvale, 2015-09-22:
chsw -> Christian Swahn
LVale -> Luis Vale

:)

Clooney82, 2015-09-22:
@fnoopdogg ? Your name?

fnoopdogg, 2015-09-23:
Fnoop Dogg :)  But my contributions are negligable, I shouldn't be included in this.

wolkstein, 2015-09-23:
hi luis,
```Anything that you guys see appropriate for licensing along the lines of GNU is fine by me```
that mean CC for your snd-files and images is ok?
if yes, with or without commercial use? https://creativecommons.org/choose/

maybe we create an license wiki site. there we can inform about all licenses we use in this project.

for exampel

main code: (GPL3)
S.Port Library: (2014) copyright by  Pawelsky. Not for commercial use!
FastLed ???
MavLink: (2009) copyright by Lorenz Meier under LGPL

media:
 Pixhawk MavLink to FrSky S.Port Converter snd-files, images, descriptions, circuit diagrams are licensed under Creative Commons License.
[image]

/g
wolke

lvale, 2015-09-23:
In the spirit of the project 

[image]<br />This work is licensed under a <a rel="license" href="http://creativecommons.org/licenses/by-nc-sa/4.0/">Creative Commons Attribution-NonCommercial-ShareAlike 4.0 International License</a>.

wolkstein, 2015-09-23:
super,
i also create an request on http://diydrones.com/forum/topics/amp-to-frsky-x8r-sport-converter to all people or developer which create media files in past, which we maybe currently use.
/g
wolke

wolkstein, 2015-09-23:
@lvale,
did you also create sonds under DisplayApmPosition/SOUNDS/en/SYSTEM ?
or are this sounds from opentx sound pack?

lvale, 2015-09-23:
All sounds were created by me, including the standard OpenTX sounds, so that the voice would be the same. Also as small suggestion, I would suggest that auxiliary files, like sounds and graphics, would preferably be named in Caps to easily distinguish them from the system ones.

lvale, 2015-09-23:
I forgot to mention, but those sounds are for OpenTX 2.0.x version only. OpenTX 2.1.x has changed somewhat the sounds.

xros, 2015-09-24:
I watched youtube. And found some guys selling this pre-programmed teensy converter at 35 English Pounds which is more than 53 US bucks.
The shop link is here: http://www.droneshop.biz/taranis-mavlink-apm-pixhawk-telemetry-adapter.html

lvale, 2015-09-24:
or https://www.airborneprojects.com/product/apm-mavlink-to-frsky-smartport-converter/

Clooney82, 2015-09-24:
I know both, but as I found out
* droneshop uses old code think from @wolkstein or @lvale and support only opentx2.0.x
* airborne use other teensy code, lua don´t know.

I think currently we are the only project which supports opentx2.1.x
And with currenty used FrSky library it is not possible to sell a commercial product with this codebase.

And because we are currently in detailed clarification, under which licence which codepart is. It should be clear for all who will use this.

xros, 2015-09-24:
@Clooney82 
airborne uses other codes as they claimed: 

http://community.emlid.com/t/mavlink-to-frsky-smartport-converter/758

They used Arduino nano for instead rather than the more expensive teensy. And their codes are opensource. But we do not know how they re-designed the Arduino nano board or not. And that is not maintained too often.

fnoopdogg, 2015-09-24:
The core teensy code is a little problematic - although Rolf stated that his project was derived from https://code.google.com/p/telemetry-convert/, his code isn't obviously a direct continuation of that project enough that it can also be considered to be covered by that GPL license.  I've messaged Rolf through diydrones to ask his intention of the code, but I haven't seen him active for a long time.  I suspect that as he released his code publicly without license, he probably meant it to be public domain (ala djbdns), but that's not good enough - it must be explicitly stated:
 http://www.gnu.org/licenses/license-list.en.html#NoLicense

The S.Port Library being proprietary is a problem as well.  That and having a further mishmash of GPL and CC licenses, this project is a bit of a mess, licensing wise :(

Clooney82, 2015-09-24:
S.Port Lib is not proprietary, only not for commercial use.

Clooney82, 2015-09-28:
Think I found a solution for our library issue: http://www.gnu.org/licenses/gpl-faq.html#GPLIncompatibleLibs

>Copyright (C) [years] [name of copyright holder]
>
>This program is free software; you can redistribute it and/or modify it under the terms of the GNU General Public License as published by the Free Software Foundation; either version 3 of the License, or (at your option) any later version.
>
>This program is distributed in the hope that it will be useful, but WITHOUT ANY WARRANTY; without even the implied warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the GNU General Public License for more details.
>
>You should have received a copy of the GNU General Public License along with this program; if not, see <http://www.gnu.org/licenses>.
>
>**Additional permission under GNU GPL version 3 section 7**
>
>If you modify this Program, or any covered work, by linking or combining it with **FrSkySportTelemetry library** (or a modified version of that library), containing parts covered by the terms of **FrSkySportTelemetry library**, the licensors of this Program grant you additional permission to convey the resulting work. {Corresponding Source for a non-source form of such a combination shall include the source code for the parts of **FrSkySportTelemetry library** used as well as that of the covered work.}

lvale, 2015-09-29:
With the addition of the link to this project on ArduPilot's wiki, I've been asked what license we use. Have we a consensus on this, so that I can report back ?
https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/126#issuecomment-143959528

Clooney82, 2015-09-29:
Code is from my side clear, only img and snd. But this should you decide @lvale

lvale, 2015-09-29:
So, any objections of having this license for the entire project??

I prefer to exclude the commercial option, so that, like was said above, we aren't compelled to provide support to commercial solutions.

[image]<br />This work is licensed under a <a rel="license" href="http://creativecommons.org/licenses/by-nc-sa/4.0/">Creative Commons Attribution-NonCommercial-ShareAlike 4.0 International License</a>.

Clooney82, 2015-09-29:
Code must be GPLv3.

SND-FIle & Images could is your decision

fnoopdogg, 2015-09-29:
"Code must be GPLv3." <-- why is that?

Clooney82, 2015-09-29:
simple, see your comment first comment in this issue.
and code should not be a CC License shouldn´t  be used for software as I know.

lvale, 2015-09-29:
Resuming
Code: GPL v3
Accessory files (Sound and Images) CC

[image]<br />This work is licensed under a <a rel="license" href="http://creativecommons.org/licenses/by-nc-sa/4.0/">Creative Commons Attribution-NonCommercial-ShareAlike 4.0 International License</a>.

Right ??

fnoopdogg, 2015-09-29:
.. which I followed up by saying that it's Rolf's code and he has to put it into the public domain, or agree to license it by GNU type license.  This is a community project now and it's just my personal opinion that I don't think it should be up to any one person to make unilateral decisions (which is one reason I suggested it should move to a community github).  Anyway, I'm a tiny part of this and I've had my say.

hamishwillee, 2015-09-29:
Just for my interest - why "non commercial use"? I'm a bit of a fan of LGPL style licenses so that people can include these ideas in their products if they want.

lvale, 2015-09-30:
On my part I tend to help everyone that asks for help on this project and other projects I'm involved.
I only get pissed when buyers from commercial products private message me "requiring" my support for something they've bought, because, unfortunately, my name is on the code. And I had too many of these....

Clooney82, 2015-09-30:
@lvale currently this project cannot be used for commercial product. Because S.Port Library.
So we shouldn´t get support requests from that way.

So, to end this discussion I will close this issue resuming:

Code: GPL v3
Accessory files (Sound and Images) CC

[image]<br />This work is licensed under a <a rel="license" href="http://creativecommons.org/licenses/by-nc-sa/4.0/">Creative Commons Attribution-NonCommercial-ShareAlike 4.0 International License</a>.

If anyone has a real problem with this, reopen this issue.
This evening I will commit license infos to github. I already prepared everything.

fnoopdogg, 2015-09-30:
There are a few misunderstandings about licenses and their effect here, but primarily I don't agree with taking someone else's code and relicensing it without permission, so please remove me as a 'collaborator' of this fork.  I'll note that I am in no way an expert on licenses or coding so it may not be an actual issue at all, just it makes me uncomfortable personally.

Aside from that, you guys are doing a fantastic job here, and I genuinely wish you the best of luck with the project.

Clooney82, 2015-09-30:
@wolkstein @lvale 
What do you think?

wolkstein, 2015-09-30:
@fnoopdogg,
i think on the teensy code site we are sure with gpl3 licence because the code root is at https://code.google.com/p/telemetry-convert/ which is sure gpl3. gpl3 also say that every code which come later and is based on the licensed code also have to publish under the same license and must calling the original author.  imo this mean, that at least the first release from rolf, and christian on github, is for sure under gpl3 license. so we are right with gpl3 on teensy code. s.port library is used but not under gpl or lgpl but also for sure not for commercial use. this is sure and we have to inform in our LICENSE text about this. for us, here is absolute no problem to use the s.port library and also to hold them in our repo.
of course because this sport lib, this project can not used for commercial use. else if the company which will do this do a deal with the sport library creator. but this is complete independent from us. we are not responsible for this. at least because we hold this information in our license text.

what s left? only the question for media files. lvale do the work on snd-files so he is the cc owner for this. some people draw circuits which we publish here and where we do not know who draw them.
so imo, we can redraw own circuits and than we are fine. beside the circuit diagrams most lua telm1 pictures are from me. i also agree to put them under cc. most text in wiki is from lvale me and clooney. so imo we also sure on this site.

closed 2015-10-04 by Clooney82