Clooney82/MavLink_FrSkySPort #31: FrSky/OpenTx to Ground Station as a relay

xros, 2015-09-24:
I saw that the Rx on teensy 3.1 board has not been used yet. 
The teensy board is only used to translate MavLink protocol to FrSky OpenTx.  It's only one direction. 
If you want to control the drone, you will have to use your RC radio and hit the stick or something on it. And here it has nothing to do with the Ground Station software.

Here I ask that how about connecting FrSky/OpenTx to ground station as a relay. 
Then we do not need 3DR digital transmitter anymore. And only one radio would work. 

How do you think?

xros, 2015-09-24:
I have not known how to pass data from opentx to another yet due to that I have not received my first FrSky radio yet :P .

Can we pass data from FrSky's serial ports to Ground station software. 
There should be two procedures: 

1) the data in mavlink protocol parsed on FrSky radio and shown on LED screen.

2) the data in mavlink protocol passed to Ground station using serial ports and keep it original 

--
Why?
--
Sometimes if you fly a drone/plane too long, the ground station is more powerful and less intensive.

xros, 2015-09-24:
P.S.
Is teensy close source? ( https://www.pjrc.com/store/ic_mini54_tqfp.html Teensy is shipped with a pre-programmed chip, and it is born close source ... )

Can we use Arduino Nano or others for instead?

Thanks!

Clooney82, 2015-09-24:
Currently it is not planned to make a Ground Relay station out of this project.
I also think it is not possible to send mavlink data from Radio to RX.

Basically old Version supported Arduino Pro Mini and teensy 2, but only with reduced functionality due to limited hardware resources of Arduino boards.

If you want a cheaper HW, you could also use a teensy-LC.

xros, 2015-09-27:
For the receiver X8R, its telemetry port is a SmartPort which include 3 Pins those are GND, Vcc, Signal.

It seems to be on direction:  FlightControllerBoard -> Receiver on pin Signal.

If we can use an extra channel to pass data to the receiver and hook that channel Pin onto the converter board which is Rx. Then the converter board can use another pair of Tx to pass MavLink data directly to the Telemetry port Rx on APM/Pixhawk. Then the path is in a circle.

Opposite direction:  GND Station + USB ->  FrSky RC controller -> Receiver -> Converter ->  Telemetry Ports on APM/Pixhawk

So I think for the relay, the full path is like:

GND Station <--> APM
Using:
GND Station <--> USB <-->  FrSky RC controller <--> Receiver <--> Converter <-->  Telemetry Ports on APM/Pixhawk

How would you think? :thought_balloon: 

:smiley: 
xros

Clooney82, 2015-09-28:
You can maybe do so, but it is not in scope of this project.

closed 2015-09-28 by Clooney82