Clooney82/MavLink_FrSkySPort #33: Telemetry using Arduino Pro Mini/Mega2560 R3 not working

xros, 2015-09-27:
I haven't received my teensy 3.1 board yet. So I picked the arduino boards which I had.

And after compiling, I found the SRAM was used up for Arduino Pro Mini.
Then I used another board to do the testing, Arduino Mega2560 R3 which has more SRAM. ( I was not planning on flying a Mega2560 to the sky,just a test :smile: )

Yes, I successfully burned the telemetry [codes](https://github.com/Clooney82/MavLink_FrSkySPort/tree/s-c-l-v-rc-opentx2.1/MavLink_FrSkySPort_Teensy2_and_ProMini) for the teensy2.0 or Arduino Pro Mini on Arduino Mega2560 R3. ( It's on my Taranis X9D Plus with opentx 2.1.3 )

And I used the Serial1 ports for telemetry and the default Serial ports for debugging.

But I could not get parameters from debugging console. Yet there were no things passing to my FrSkyTaranis X9D Plus RC controller but the RSSI, SWR, Receiver battery voltage. 

I don't know whether the MavLink protocol was translated on the board.

1) Do you know how to fix it?

2) Or if for arduino Pro Mini, which scope of codes is compatible, can you provide the link?

Thank you! :bow: 

xros

Clooney82, 2015-09-28:
As you see the code for teensy2 or adruino pro mini is now over 6 month old, and I don´t hava a arduino I can´t make any suggestions

Clooney82, 2015-09-29:
Will close this because of unsupported HW.
You still can comment if needed.

closed 2015-09-29 by Clooney82