Clooney82/MavLink_FrSkySPort #34: Support of other hardware than Teensy 3.x, LC

Clooney82, 2015-09-28:
Will we still support other hardware than Teensy 3.x, LC.
From my side I cannot do this.
So either we drop support or we need someone who will maintain code for Teensy2 and Arduino ProMini.

lvale, 2015-09-28:
It's the same problem that made ArduPilot drop the support for the APM boards. If Arduinos and such would have to be supported the effort to have code running on it would increase significantly, and certainly lots of features would be impossible to accommodate and would also limit future developments. The old argument about the cost of the Teensy 3.x vs the Arduinos is IMHO a moot point, because people are using 200 euros + radios with hundreds of euros vehicles so the cost of 20 euros vs 5 is a false question. So, I would say to focus on the target platforms, and drop the rest.

wolkstein, 2015-09-28:
no,
only teensy3.X and LC.
LC also can only use adc with 12bit resolution. to measure 6s not so precise than 13bit.

xros, 2015-09-28:
BTW, this is the arduino board [arduino sparkfun pro micro](https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide) which has a higher SRAM because it has an ATmega32U4 core. And it works for the project at the present but not future guaranteed.

wolkstein, 2015-09-28:
32U4 is used also by teensy2 and also have not enough memory for all messages. it makes absolute no sense to waste time with other boards than tennsy3.x versions if you want to have an nice converter. i use an teensy2 some month ago because my teensy3 was burned. if i am waiting for the new teensy3 i create this teensy and pro mini port. https://github.com/wolkstein/MavLink_FrSkySPort/tree/s-c-l-v-rc-opentx2.1/MavLink_FrSkySPort_Teensy2_and_ProMini
it works well with opentx 2.0.17 and non EU x8r receiver. please use this if you need teensy 2 or pro mini support. but since we start this project we also decide some weeks ago on diydrones forum that we will only support teensy3.x boards!
/g
wolke

closed 2015-09-29 by Clooney82