Clooney82/MavLink_FrSkySPort #35: Nativ Mavlink Telemetry in OpenTX

dk7xe, 2015-09-28:
No issue but request for help pushing the request on the opentx branch ;-)

There is request [#2354](https://github.com/opentx/opentx/issues/2354) on the opentx branch requesting native mavlink support on opentx.
Means that opentx does the mavlink parsing and values are available as telemetry.

That would eliminate the need for an Teensy and the small size [FUL-1](http://copter.ardupilot.com/wiki/common-frsky-telemetry/) can be used for people not needing single cell voltage/LED control.

On the other hand the Teensy does not need to translate Mavlink into FrSky but just parse MavLink and do packet/data injection if needed eg. for single cell voltage.

if you also see benefits please push the [#2354](https://github.com/opentx/opentx/issues/2354) request by adding comments there.

Thank you!

Clooney82, 2015-09-28:
Would be great to have. But the question is how to send MAVLink Data to RC.
I only now about openLRS implementations with working MavLink support.

dk7xe, 2015-09-28:
#2354 is about having the same stuff than openLRS at opentx. if more of us are leaving a note at #2354 the chance is higher that we get it.

xros, 2015-09-28:
This is very exciting. I posed a comment [here](https://github.com/Clooney82/MavLink_FrSkySPort/issues/31) . But this talk was closed by @Clooney82  . In the 2nd comment, procedure  2, if we can pass the native MavLink protocol to the RC, then the ground station relay is possible via serial ports.

lvale, 2015-09-28:
@xros, as @Clooney82 said, it's not the scope of this project for now. You can pursue that possibility, but the Taranis and Lua is not the appropriate way to build a Ground Station (at least not without heavy modification of the base code - OpenTX)

closed 2015-10-07 by Clooney82