Clooney82/MavLink_FrSkySPort #36: Copter 3.3 Final in on

lvale, 2015-09-29:
Hi

Just checked that 3.3 Copter is released as a soft release. It won+t appear for a few days yet on the GCS menus, but if one chooses beta firmwares the final code is available.

Clooney82, 2015-09-29:
So we need to get the new status messages.
Everything else should work. I´m using AC3.3 since rc5.

Clooney82, 2015-10-02:
Just greped hopefully all Statustexts from Ardupilot 
Can someone please verify this?
See commit: https://github.com/Clooney82/MavLink_FrSkySPort/commit/c7687d36b18c16d6a07836f21a467854b6904a74

Clooney82, 2015-10-04:
Would be great if someone could test this: https://github.com/Clooney82/MavLink_FrSkySPort/commit/5fa06f3a2750917486ee74a856e7db38eed87304
https://github.com/Clooney82/MavLink_FrSkySPort/tree/statustext_AC3.3

wolkstein, 2015-10-05:
is copter 3.3 really stable?
i plan switch from 3.2 to 3.3. too also make contributions here. but i am unsure if i still wait for next bugfix-release. i need a real secure system because my pixhawk fly in a commercial used copter.
what do you think?

Clooney82, 2015-10-05:
You will currently only get it only via beta firmware, but it is no longer marked as RC.
It will be moved to stable soon. see: http://diydrones.com/xn/detail/705844:Comment:2097154
```
APM:Copter Release Notes:
------------------------------------------------------------------
Copter 3.3 29-Sep-2015 / 3.3-rc12 22-Sep-2015
``` 
I use it since rc5, last flights were with rc11, for testing latest version I had no good weather till now.
Untill now, no real problems, pretty good flying with my V-Tail.

But to be save you could download AC3.2.1 FW and save it, to downgrad if needed.

lvale, 2015-10-05:
I've been testing the RC's and the latest RC12 is as "clean" and good as you can get from the targets that were set for 3.3, so that it was marked as final. Randy decided to do a soft release for the 3.3 so that users could check further, but nothing relevant has come up so far.
When I had some good weather here I managed to get some videos:

https://youtu.be/ZqnzeQ-LLTk
https://youtu.be/J8yJ5_1Xyrg

Please note that I always do a clean install of every build, so I always have to recalibrate everything. Testing was done on both a 3DR PixHawk and on the videos is a PixHawkMini from GLB.

Please become used to the new settings for CH7-CH12 (which were expanded from CH7-8) and the new options like Emergency Stop and Motor Interlock. Also some new parameters grant some testing and exploration on the machines, specially 

MOT_THR_MIX_MAX,
MOT_THR_MIX_MIN,
MOT_THST_BAT_MAX,
MOT_THST_BAT_MIN,
MOT_THST_EXPO,
MOT_THST_MAX,

Clooney82, 2015-10-06:
Would be great if someone could verify my last commit ( https://github.com/Clooney82/MavLink_FrSkySPort/commit/81aafee4fc358055e5df66bb97af42b9bc4d6ed2 ) to branch statustext_AC3.3.
Hopefully I got now all messages, except dynamic ones.

closed 2015-10-11 by Clooney82

Clooney82, 2015-10-15:
@lvale @wolkstein 
Do you think we can release a new Version ?