Clooney82/MavLink_FrSkySPort #37: Compile for Teensy LC

Bingisser, 2015-10-03:
Hello,
I tried to compile "s-c-l-v-rc-opentx2.1"  for "Teensy LC" but I ran in problems, I got: 
C:\Program Files (x86)\Arduino\libraries\FrSkySportTelemetry/FrSkySportSingleWireSerial.h:12:28: fatal error: SoftwareSerial.h: No such file or directory
#include "SoftwareSerial.h"

Pawelsky wrote on http://www.rcgroups.com/forums/showthread.php?t=2245978  "a library, that allows to emulate the FrSky S-Port sensors using Arduino compatible Teensy 3.x board or 328P based boards .." does this library works for Teensy LC?
I tried al lot of thing, new Install Arduino, to include the SoftwareSerial library in main sketch, no change. What can I, a Noob, do?

Clooney82, 2015-10-03:
Did you installed all software described in our [Wiki](https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.-Setup)?

SoftwareSerial should be included either with arduino installation itself of via installing teensyduino.
Please check if Library is installed.
Sketch -> Include Library -> SoftwareSerial
or
Sketch -> Include Library -> Manage Libraries -> Search: SoftwareSerial

Clooney82, 2015-10-06:
@Bingisser does it work now?

Bingisser, 2015-10-06:
Hallo Jochen,
Ich bin im Moment gerade einige Tage auf Städtereise in Hamburg ;)

Ich habe auf 2 PCs mehrmals zuerst das neuste Arduino, danach die Teensy Software installiert. Das Kompilieren für Teensy 3.2/3.1 und 3.0 funktionierte jeweils tadellos! Einfach nicht für LC.
So wie ich es im Kopf habe ist SoftwareSerial installiert. Ich prüfe es noch sobald ich zurück bin. Gibt es mit dieser Library nicht neustens eine Hardwareabhängigkeit, welche in früheren Arduino Versionen nicht bestand? 

Gruss und besten dank für die Nachfrage,
Jürg

Clooney82, 2015-10-06:
Hi Jürg,

sollte es wider erwarten nicht gehen, muss ich den LC wohl wieder von der Liste streichen.
Den Daten nach sollte es aber eigentlich gehen.

Clooney82, 2015-10-07:
Kopf -> Tisch !!!
Eigentlich benötigen wir für die Teensy SoftwareSerial garnicht.
Wird auch so von der FrSkySportTelemetry Library abgefragt, leider wird darin der LC aber nicht erkannt und greift somit auf SoftwareSerial zurück, was natürlich nicht geht, da wir keine PINs für SoftwareSerial definiert haben.
Dachte ich hätte das schon angepasst, hab ich aber wohl vergessen. 
@Bingisser : Versuch mal folgendes:
Editiere die Dateien **FrSkySportSingleWireSerial.h** & **FrSkySportSingleWireSerial.cpp**

Und suche & ersetze folgende Einträge:
1. Suche:
```
#if !defined(__MK20DX128__) && !defined(__MK20DX256__) 
```
und ersetze durch folgendes:
```
#if !defined(__MK20DX128__) && !defined(__MK20DX256__) && !defined(__MKL26Z64__)
```

2. Suche:
```
#if defined(__MK20DX128__) || defined(__MK20DX256__)
```
und ersetze durch folgendes:
```
#if defined(__MK20DX128__) || defined(__MK20DX256__) || defined(__MKL26Z64__)
```

Kann es aktuell leider nicht selbst testen, da sich mein LC in Rauch aufgelöst hat und ich noch keinen neuen bestellt habe.

/g Jochen

Clooney82, 2015-10-10:
@Bingisser: Gibt es schon was neues?

Bingisser, 2015-10-10:
Hallo Jochen,
Konnte heute nachmittag die Anpassungen gemäss deinen Angaben machen, nun funktioniert das Kompilieren wie gewünscht!

besten Dank für die Hilfe und die Top-Arbeit!
Jürg

Clooney82, 2015-10-10:
Danke, sehr gut.

commit 05d305a2 (2015-10-10): Fix compile issues with Teensy LC fix / close #37
closed 2015-10-10 by Clooney82