Clooney82/MavLink_FrSkySPort #39: Lipo Singe Cell Monitor - write down formula for divider and don't use custom values in example

robert-budde, 2015-10-04:
Hi!

Great project!

I was irritated by the posted values in the wiki for the divider. As for people who don't wnat to take a closer look into the code this is just a black box, the equation leading to the divider should be written down somewhere. After figuring out ADCref ist 3,3V and 8191/3,3 * Rgnd / (Rseries + Rgnd) it still does not match the values - as the values in the wiki are given for 0,001% accuracy (really!?) this is misleading.

BR
Robert

wolkstein, 2015-10-04:
hi, resistors firstly are calculated against 1m output resistance. this mean for example 4,2 v (1s) on 1000ohm series and 3300ohm to gnd result in 3.220785 Volt. this is in the theory. all default 13 bit software dividers not calculated in theory. they are calculated with an existing resistor network (my 6s network). i use the resistors which are shown in wiki table. i use simple metal film resistors with 5% tolerance from e24 series. this mean as shown for 1S the calculation is not made to reach 3.3V. it is made to reach a near by value as high as possible with the use of e24 series resistors.
second. board voltage is simple on each set-up different. i use a castle bec with 4.9V to power my FC and we use teensy default reference voltage, and not an external exact ADCref. dependent to users board voltage 4.7-5.6V and teensy simple 3.3 voltage regulation with also have tolerances, it is not possible to give a default precision from 0.01V at each cell. each user who want to have this super precision (including me) have the need to calculate his own corrected dividers for his existing environment.

and for sure i now this problem, i also was thinking about an simple calibration procedure. currently this was moved because we was working on other things to release our first stable version 1.6.

this is the reason why no concrete  infos about calibration are written in wiki. also people who are able to build there own network and do the solder job may also advanced enough to find out how the code behind work and also can perform an manual calibration/software divider calculation.
if we create an simple calibration method we(I am) also will finished the dependent wiki page.

gnkarn, 2015-10-04:
you may find this excel calculation and explanation useful ,
( it seams i can't attach an excel file , is there other way ?)

gnkarn, 2015-10-31:
BUG  it seams there is an important bug on the cell voltage calculation.
to simulate the conditions just put everything to work normally,  then disconnect just the lipo balance connector .
the cells and zelle calculations will remain at the previous values, even when , read is 0.

wolkstein, 2015-11-01:
if you disconnect the balancer plug the teensy adcs inputs are not zero. they values randomly change in this case.  because for secure reason the gnd line from resistor network is only connected to balancer plug gnd and not to teensy. that mean if you disconnect the balancer plug the adc inputs are simply open. see schematic here  https://github.com/Clooney82/MavLink_FrSkySPort/wiki/2.1.-Lipo-Singe-Cell-Monitor .

!!! please do not change this and loop the resistor network gnd to teensy gnd. this can damage the whole electronics because current can flow through gnd if your lipo-main-plug (+) is connected while lipo-main-plug (-) is disconnected but the balancer plug is already connected to resistor network !!!

/g
wolke

gnkarn, 2015-11-01:
That is not on my case .
I fully understant what you mean . But in my design All is grounded and the
adc  properly shows zero .

The problem is that the logic is not clearing the previous values . It
could be that the algorithm is detecting a a no cell . Because the first
check for i=0 is 0 . Then it never clears the cells values .

It is easy to simulate .
Just try ..

Gus

wolkstein, 2015-11-01:
ok,understand. at least the cell detection can be removed. it makes no sense at all and users simply define "maxcells" more or less as "cells". so the detection makes no sense and than valus less than _analogread_threshold (default 100) will measured.