Clooney82/MavLink_FrSkySPort #42: include a user defined function

gnkarn, 2015-10-13:
in order to make the code transparent to accommodate a user defined function without the need to include it on the main code , it could be of great help to add into the main LOOP , a user_definedFunction() ; line of code .
and also include a void user_definedFuncition{} ; separate file for the user to complete ( if needed)

Clooney82, 2015-10-16:
Hi,
in my opinion, this is not needed.
Cause main loop is only about 30lines of code (Line 376 - 408).
It should be easy for all people who has some knowledge of programming in arduino.

Maybe the other devs has a other opinion.

I hope you can understand my view.

lvale, 2015-10-16:
I agree with @Clooney82 
The script code initially was (and still is) divided in functional modules (GPS Panel / Top Panel, etc) so that these could be removed if needed or replaced with better or other options.
The main loop is a result of that approach and can easily be extended.

Take in consideration that there are severe restrictions of what the script can execute.

gnkarn, 2015-10-16:
Let me defend my proposal . 
In my case I have added a front flash led with a simple  coded blinking pattern .
It would be useful to be able to update  the code without the need to modify anything on the main code. 

By just adding a user function in the main loop and a user setup on the setup.code .

The cost for this is cero. And does not involve Lua scripting.

Thanks

lvale, 2015-10-17:
You'll have to recompile the code, but you do already have a feature that does what you want. 

Just use the current file

https://github.com/Clooney82/MavLink_FrSkySPort/blob/s-c-l-v-rc-opentx2.1/MavLink_FrSkySPort/led_control.ino

to do what you want.

gnkarn, 2015-10-17:
Never mind . It was to provide a way for othera to include their own functions easily .
I'm developing my own code . I m very interested on the code you guys developed . I'm  wing using also fasteled for a year now , and in my case the led I have added is a regular led connected to an analog output and I'm using fasted functions to manage it .

I would share it but don't know how to do it on git.
Thanks

lvale, 2015-10-17:
@gnkarn 
No problem. If you think it might be useful for others, it can either be pushed to the main branch or create a new branch. I'm sure that @Clooney82 won't mind if you send him a zip file with the code so that he could do it.

closed 2015-10-23 by Clooney82