Clooney82/MavLink_FrSkySPort #43: Distance from home not displaying in radar

athertop, 2015-10-15:
Running latest 2.1 code (and with APM 3.3)
distance from home - the arrow on the radar display, even though it pointed in the correct direction for the copter heading, it stayed in the centre and the distance value under this radar stays at '0m'.

wolkstein, 2015-10-22:
did you use the latest rev of lua script. this issue is fixed in https://github.com/Clooney82/MavLink_FrSkySPort/commit/56e56d33d5f0aabcea9d29e0e3ea2f520982219e

/g
wolke

Clooney82, 2015-10-30:
@athertop is this still a valid issue? or is it solved?

athertop, 2015-10-30:
@Clooney82 I have been unable to test due to the telem screen memory issues causing the screen to blank out. If the memory issues are sorted in the latest -next version then I can hopefully test this over the weekend. Can you confirm status of memory issues please?

Clooney82, 2015-10-31:
@athertop, so this is only next branch related?

athertop, 2015-10-31:
No, it's related to the release version which was current at the time I raised this. Not sure if it's resolved in the present release version now,  as I have -next branch now installed so unable to test current release version

wolkstein, 2015-11-05:
hi, the stable version of telemetry script have mostly no memory problem. i rewrite this version from laves version with helpfully tips from opentx developer to save memory as possible and keep the garbage collector against to much work.  for example because that, i do not use dynamic objects and also make all vars script wide and not in the normal way as local function vars. also it is important that all vars, "must" initialised at script start to prevent mostly against dynamic memory allocation problems if memory use is at the limit.
@ Clooney82, removing comments from lua scripts do not change the memory problems.
see https://github.com/opentx/opentx/issues/1966 and https://github.com/opentx/opentx/issues/1870 and https://github.com/opentx/opentx/issues/1834. or search in opentx issues with commenter:wolkstein .

second, i use the stable script and since i update to apm3.3 i also have also the problem that the radar do not work as expected. some times it works but mostly not. think this is a problem because armed status is now not simple 1 or 0. i noticed that the flight time timer works as expected. so i am a little bit confused here why the preamheading which is needed from radar is not so clear. i think the problem is that prearmheading and pilotlat&pilotlon are overwritten because wired armed status.

/g
wolke

athertop, 2015-12-24:
Wolke,
I just setup a brand new quad with new teensy 3.2 (running AUAV-X2 FC) and have installed the 2.1 branch (not the -next branch Jochen is working on), and All seems to work OK, except for the radar. It does not work at all - the arrow stays centred and the distance reading remains on 0m. Can you please advise is this a known issue still, or one specific to me and my configuration? Thanks, Paul

athertop, 2015-12-30:
Update - issue resolved!
I managed to figure out a resolution to this issue - it is not caused by the code at all, but rather by the way the GPS sensor is added within OpenTx. I originally configured the sensors manually as per the Wiki  but adding a GPS sensor manually does not configure the sensor properly on the Taranis radio. A manually added GPS sensor will display 5-6 numeric digits instead of a Raw GPS reading, and the LUA script does not like this! Instead, the GPS sensor needs to be 'discovered' in the telemetry page on the Taranis. Doing it this way formats the sensor in a custom (GPS kind of) way.

I think it would be a great idea if the Wiki was updated to reflect this potential pitfall - Maybe add a couple of lines to explain that the suggested way to setup all the sensors is something like this:
First (with no sensors configured) do a discovery, where all the discovered sensors from the Teensy will be added; and next, edit any sensors which need to have names changed, such as Temp1 -> T1 and Temp2 -> T2, then add calculated sensors, and we're done!. Following this approach ensures that the GPS sensor is added correctly.
Thanks, Paul

closed 2015-12-30 by athertop