Clooney82/MavLink_FrSkySPort #45: void LSCM::setCellDivider() isn't correct!

darkcheater, 2015-11-04:
I was wondering why just the last cell was displaying the right value and started debugging.
I found out that just the last value of the individualcelldivider was taken from my function call in MavLink_FrSkySPort.
When looking at the setCellDivider function, the fault is clear:
The if condition becomes true just for the last cell value.

I made a fix:

[code]void LSCM::setCellDivider(double a, double b, double c, double d, double e, double f,
                          double g, double h, double i, double j, double k, double l)
{
  if(a==0.0)
    return;
  _individualcelldivider[0] = a;
  
  if(b==0.0)
    return;
  _individualcelldivider[1] = b;
   
  if(c==0.0)
    return;
  _individualcelldivider[2] = c;
  
  if(d==0.0)
    return;
  _individualcelldivider[3] = d;
  
  if(e==0.0)
    return;
  _individualcelldivider[4] = e;
  
  if(f==0.0)
    return;
  _individualcelldivider[5] = f;
  
  if(g==0.0)
    return;
  _individualcelldivider[6] = g;
  
  if(h==0.0)
    return;
  _individualcelldivider[7] = h;
  
  if(i==0.0)
    return;
  _individualcelldivider[8] = i;
  
  if(j==0.0)
    return;
  _individualcelldivider[9] = j;
  
  if(k==0.0)
    return;
  _individualcelldivider[10] = k;
  
  if(l==0.0)
    return;
  _individualcelldivider[11] = l;  
    
}[/code]

Clooney82, 2015-11-05:
@wolkstein

gnkarn, 2015-11-05:
Have you implemented the resistor network on your teensy ?

-------- Mensaje original --------

Fecha: 04/11/2015  18:50  (GMT-03:00) 
A: Clooney82/MavLink_FrSkySPort <MavLink_FrSkySPort@noreply.github.com> 
Asunto: [MavLink_FrSkySPort] void LSCM::setCellDivider() isn't correct! (#45)

darkcheater, 2015-11-05:
What do you mean by that?
The setCellDivider function is broken, as I explained in the first post.

You can either set the values in LSCM.h or use my fix and set the values in MavLink_FrSkySPort.ino as it is designated by the coders.

I personally don't know why we need such a fancy set function for that, as the divider values are just set once..
The old method (just hardcoding the array) was more efficient and needed less space, I guess.

wolkstein, 2015-11-05:
that is true, but i also planned some extras in lscm class. for example to make it possible to use it with resistor networks designed for parallel connected cells. and also i plan to create an simple calibration method. at least the class is at the beginning.
for me it is important to make the class fully configurable from main code settings. so this mean setting cell dividers by calling a function is more class orientated  as changing class headers files to make the class working for you.

thanks for the bug report. i will change this soon.

@gnkarn, i create the cell monitoring fork last year. also i create the resistor network. it is simple and not protected against all user mistakes. it is important that you now what you do if you use it. and also use it all time carefully.
since i use it, one teensy become a big smoker and also, because i have to 3s lipos in series, i burned three times my balancer cables because i do a mirrored connection.

darkcheater, 2015-11-05:
Thanks for explanation. For enhancement plans, a class based realisation is of course always a good base.
Thanks for changing.

wolkstein, 2015-11-09:
fixed with commit 30b99be6e4299554c076b36310675e61d2a9c0a1.

closed 2015-11-09 by wolkstein