Clooney82/MavLink_FrSkySPort #46: opentx2.1.5 script error because less memory

wolkstein, 2015-11-09:
i update my taranis today to 2.1.5.
now my telm1.lua stops with less memory error.

with 2.1.3 and the same setup i have no issues.

wolkstein, 2015-11-09:
wired, if i remove all comments and empty lines and reduce Indent from 8 characters to 1 characters the scripts run.
puh, normally the lua interpreter must ignore all comments and spaces.

Clooney82, 2015-11-09:
I recognized the same with open-tx 2.1.3
Maybe it is possible to simplify some parts in the main script or reducing amount of mixer scripts.
Currently I´m figuring out how to store data in files on SD-Card and remove all config mixer scripts.

wolkstein, 2015-11-09:
sd card is good. but how to parse/open a file if memory is less?

i create an issue at opentx because lua memory problem. in past and with th old 2.0.x commented text make no problems. i think they change some lua interpreter settings to make debugging possible. 2.1.15 also print advanced script error messages.

wolkstein, 2015-11-09:
hi,
i found out, if i change back minuteBeep=true ore false to minuteBeep=1 or 0  , which not work right, that the memory problem is gone and the script start.

Clooney82, 2015-11-10:
strange, cause devs said that wrong, and we must use true and false

wolkstein, 2015-11-10:
yes strange. did you test the minuteBeep thing  on your taranis?

Clooney82, 2015-11-10:
On mine it is working. 3072b free.
But I have no complex model setup.
Mine is identical to the model we provide.
Maybe some configs need more memory.

wolkstein, 2015-11-10:
ok, mine also say 3072b free independent from if i have a crash or not. additional to our three mixer script  i use a fourth mixer script for pilot gimbal tilt control.
if i remove this script it our scripts work without effects. even if use true or false in minute beep. but curios also i get  the same 3072b free mem.
think all this mean we are at the memory limit.

AffeP, 2015-11-15:
Got the same on 2.1.6, but if i go in and change telemetry screen to something else and than change it back to the script it will run OK.

athertop, 2016-05-30:
I find that if I have the wrong model selected on powering the Taranis (and this initial model is also running the lua script), then I select the correct model and go to the telemetry page, I still get a memory error - its as if that original script (from the initial model) has not been fully released from memory. The solution for me is to shut down and restart the taranis with the correct model selected at boot, and I can load the plane217/copter217.lua without memory issues. I am presently running 2.1.8.

@wolkstein - you still seeing the memory errors on the latest OpenTX? (2.1.8)?

athertop, 2016-09-11:
I am no longer seeing this issue with OpenTX 2.1.8. Closing

closed 2016-09-11 by athertop