Clooney82/MavLink_FrSkySPort #47: error messages on taranis

beb38138, 2015-11-13:
when i arm my apm 2.6, i get two messages read aloud from the taranis. they are Arm throttle below failsafe, and Arm leaning. Then i get the arming motors/motors armed. I dont see these messages in the Mission Planner so im wondering if its an error in the lua script maybe? Using 2.1.5 open tx and latest version apm 2.6

wolkstein, 2015-11-14:
hi, did you also update your teensy? new apm3.3 status messages are also updated in teensy message to id parser. https://github.com/Clooney82/MavLink_FrSkySPort/blob/s-c-l-v-rc-opentx2.1/MavLink_FrSkySPort/Avg_statustext_v3.3.ino

/g
wolke

beb38138, 2015-11-14:
thanks for your response. I noticed when i opened up the file (the one i had already uploaded to teensy), it has bot the v3.2 and the v3.3. Should i delete the first version or upload both? Thank you for your help!

   

wolke—

beb38138, 2015-11-14:
So i tried it with just the v3.3 and this is the error message i get....im clueless on what this means, but maybe it will be an easy fix

Arduino: 1.6.5 (Windows 7), TD: 1.25, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz optimized (overclock), US English"
Using library GCS_MAVLink in folder: C:\Program Files (x86)\Arduino\libraries\GCS_MAVLink (1.0.x format)
Using library Time in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Time 
Using library FrSkySportTelemetry in folder: C:\Program Files (x86)\Arduino\libraries\FrSkySportTelemetry (1.0.x format)

C:\Program Files (x86)\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ -c -O -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=125 -DARDUINO=10605 -DF_CPU=96000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3 -IC:\Program Files (x86)\Arduino\libraries\GCS_MAVLink -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Time -IC:\Program Files (x86)\Arduino\libraries\FrSkySportTelemetry C:\Users\Owner\AppData\Local\Temp\build2694242226788568751.tmp\LSCM.cpp -o C:\Users\Owner\AppData\Local\Temp\build2694242226788568751.tmp\LSCM.cpp.o 
C:\Program Files (x86)\Arduino/hardware/tools/arm/bin/arm-none-eabi-g++ -c -O -g -Wall -ffunction-sections -fdata-sections -MMD -nostdlib -fno-exceptions -felide-constructors -std=gnu++0x -fno-rtti -mthumb -mcpu=cortex-m4 -fsingle-precision-constant -D__MK20DX256__ -DTEENSYDUINO=125 -DARDUINO=10605 -DF_CPU=96000000 -DARDUINO_ARCH_AVR -DUSB_SERIAL -DLAYOUT_US_ENGLISH -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3 -IC:\Program Files (x86)\Arduino\libraries\GCS_MAVLink -IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Time -IC:\Program Files (x86)\Arduino\libraries\FrSkySportTelemetry C:\Users\Owner\AppData\Local\Temp\build2694242226788568751.tmp\MavLink_FrSkySPort.cpp -o C:\Users\Owner\AppData\Local\Temp\build2694242226788568751.tmp\MavLink_FrSkySPort.cpp.o 
Mavlink.ino: In function 'void _MavLink_receive()':Mavlink.ino:496:70: error: 'parseStatusText_v3_2' was not declared in this scope'parseStatusText_v3_2' was not declared in this scope      From: wolke <notifications@github.com>
 To: Clooney82/MavLink_FrSkySPort <MavLink_FrSkySPort@noreply.github.com> 
Cc: beb38138 <beb38138@yahoo.com> 
 Sent: Saturday, November 14, 2015 1:52 AM
 Subject: Re: [MavLink_FrSkySPort] error messages on taranis (#47)
   
hi, did you also update your teensy? new apm3.3 status messages are also updated in teensy message to id parser. https://github.com/Clooney82/MavLink_FrSkySPort/blob/s-c-l-v-rc-opentx2.1/MavLink_FrSkySPort/Avg_statustext_v3.3.ino/g
wolke—

Clooney82, 2015-11-14:
You need both files, but you have to adjust line 124/125 in MavLink_FrSkySPort.ino

beb38138, 2015-11-14:
BINGO!!! Thank you so much!! Im getting calibrating barometer messages but thats good. no more arm leaning or arm throttle warnings. thank you!!

closed 2015-11-17 by Clooney82