Clooney82/MavLink_FrSkySPort #48: LED Control

WeareGermans, 2015-11-22:
Hi,

i would like to use the led_control part of this code.

I did as stated in the wiki:

I copied all led_control.ino to the right folder (separately).
i uncommented USE_RC_CHANNELS and USE_TEENSY_LED_SUPPORT.

But it does not compile any more.
The compiler states that 'Teensy_LED_Init()' and "Teensy_LED_process()" in their respective main file function and all functions in "Teensy_LED_process()" and "default_mode(int, float)" of led_control "was not declared in this scope".

I use arduino ide 1.6.6 and AC 3.2 (commented accordingly).

How can i get it to compile?
Thanks very much!

Clooney82, 2015-11-24:
Hi,

how is your directory structure?

WeareGermans, 2015-11-24:
Hi,

i just downloaded the whole repository as zip file and extracted it under downloads (on Win 7 x64).
So all under C:\Users\Franz\Downloads\MavLink_FrSkySPort-s-c-l-v-rc-opentx2.1\

And then copied the respective led_control.ino up one folder overwriting the existing file.
I installed the provided Libs via the IDE Function.

WeareGermans, 2015-12-02:
Could you tell me which combination of OS, Teensy IDE and Arduino IDE does work for you?

wkhard, 2015-12-10:
Hello,
I find the exactly the same problem. With all three of the led sketches.
Tried all hints I could find in the web....
Win7, teensy 1.26, Arduino 1.6.5.
Can anybody suggest another teensy and/or Arduino Version where the compiling works?

gnkarn, 2015-12-11:
http://youtu.be/MC8WzAcPvls

IT works fine ,

Clooney82, 2015-12-11:
This is how it looks here:
[image]
[image]
Using Mac OSX 10.11.2, Arduino 1.6.5, teensy 3.1, teensy loader 1.24

WeareGermans, 2015-12-11:
Hello,
Thanks for the hint!

I deinstalled everything and reinstalled Arduino IDE 1.6.5-r5 with teensyduino 1.26 on win 7 -x64, and now it works.
It seems that your code is maybe not (yet) compatible with Arduino IDE 1.6.6.

It could be a help for others, if you collect the working combinations of the Arduino IDE and Teensy Software in the WIki.

Thank you again for help!

wkhard, 2015-12-11:
Thanks you also!
Finally I could Compiler it also.
B.r.
Kurt

closed 2016-01-17 by WeareGermans