Clooney82/MavLink_FrSkySPort #49: Not all telemetry values are shown

darkoperator, 2016-01-07:
Hi

I'm using a teensy-lc and not all telemetry values are shown on a APM 2.6 running the last supported version of Arducopter, on the Telemetry window voltage is displayed incorrectly vs what the sensor shows. HOP and satellite info is not shown, same for heading I get blank values on those. I can see mode is shown correctly and on the telemetry sensor value the voltage and accelerometer info for y/z vibration and current are shown correctly int he sensor info.

tampangsuorr, 2016-02-12:
Did you reduce the level to 3.3v between the telemetry port of the apm and the teensy lc?

athertop, 2016-03-30:
Do you have OpenTx 2.1.7 by any chance on your Taranis? If so, there is a copter217.lua file which allows the telemetry page to work correctly with the new number format changes which happened in OpenTx 2.1.7. For older versions of OpenTx 2.1 the regular telem1.lus ashould be used from the 2.1 specific version of this repository.

darkoperator, 2016-03-30:
Have not tested it will check

athertop, 2016-05-30:
@darkoperator - did you have time to test the latest lua script to see if this resolves your issue?

athertop, 2016-08-22:
No response since May. Assuming resolved. Closing.

closed 2016-08-22 by athertop

darkoperator, 2016-08-22:
yep, sorry. I retired the quad with the APM