Clooney82/MavLink_FrSkySPort #50: OpenTx 2.1.7 - new PREC1 PREC2 formatting breaks telem screen

athertop, 2016-01-13:
I did a test flight tonight using OpenTx2.1.7 for the first time. The battery and cell voltages were reading at just 10% of their actual values and satellite count was not displying correctly either. Did some research and it appears that in 2.1.7 the OpenTx devs have reverted the functionality of the PREC1 and PREC2 formatting within LUA. I'm not sure if this is the cause of the issue but it seems likely that it is. The formatting change in 2.1.7 is described here: https://github.com/opentx/opentx/issues/3150

This will require some changes to the LUA to make it comply with the new formatting changes.

athertop, 2016-01-13:
I have tested fixes and can advise that my assessment in OP was correct - With 2.1.7 to display values with PREC1 you need to multiply the source value by 10  and for PREC2 values, you need to multiply the source value by 100, then they display correctly again. My fixed telem1.lua is here for reference: https://dl.dropboxusercontent.com/u/12131631/forum-links%28do%20not%20delete%29/telem1.lua

athertop, 2016-01-18:
I have created a new copy of telem1.lua called coptr217.lua in my fork which I will generate a pull request for. It will give the user the choice of using the current telem1.lua for OpenTx versions 2.1.0-2.1.6 and for 2.1.7 the suggestion is that they can use the new coptr217.lua. I have named this with prefix of coptr as I am working on a plane version. Pull request coming.

Clooney82, 2016-01-18:
Good Idea.

athertop, 2016-01-19:
Have raised pull req for this new telem script to be added to project. To use this new coptr217.lua script on a Taranis, the user would need to copy this to the /SCRIPTS/TELEMETRY folder on their SD card and on the model telemetry page replace telem1 with coptr217 (once the 2.1.7 OpenTx firmward update has been completed of course :-)

athertop, 2016-02-14:
As the merge happened, I would consider this issue now closed. I think it would be a good idea to update the Wiki to document. I am happy to do this with your approval Clooney?

closed 2016-03-30 by athertop

Big-Iron, 2016-09-11:
Hello Paul, 

found that the coptr217.lua is missing in the s-c-l-v-rc-opentx2.1-next

athertop, 2016-09-11:
Correct! That next brach is an experimental branch taken from a much earlier revision of the code, and has never been updated, so should be treated as a stale branch now.