Clooney82/MavLink_FrSkySPort #51: Pause Mode not displaying/calling - code fix detailed in post

athertop, 2016-01-17:
If you select Pause mode (new in 3.3) it siaplays/calls "Invalid Flight mode". I have tracked this down to line 498 which reads:
if FmodeNr<1 or FmodeNr>17 then

it should read:

if FmodeNr<1 or FmodeNr>18 then

FmodeNr is set in the LUA to the Telem Flight Mode + 1, but the Pause Mode is APM mode 17, so this sets FmodeNr to 18 which the current line 498 equates to false so fmodeNr is set to 13 in the else comparator which is Invalid flight mode. Simple fix - replace 17 with 18, or instead use table.getn(FlightMode) to simply refer to the array length. This way you can add future flight modes to the start of this code without worrying about This IF statement.

lvale, 2016-01-17:
Are you sure is "Pause".

Last I've seen was "Brake"

athertop, 2016-01-17:
Well spotted - my deliberate mistake ;-) It is indeed 'Brake mode', anyway, changing 17 to 18 fixes this.Strangely, the wav file for Brake mode is already present in the repository, so we are all good there.

lvale, 2016-01-17:
Yes.
As Brake mode can be used on a Aux function (CH7 to CH12) the sound appears isolated (I use it on CH7), but for the sake of consistency can be added to the Flight modes list also.

athertop, 2016-01-17:
Ivale, The sound files for the "Brake" mode and the coding is already configured/included in the project (you will see the FlightMode array which starts on line 102 includes the "Brake" mode) - its just that the coding error was preventing them from being used. Like I said, changing 17 to 18 on line 498 fixes this. I am hoping that Clooney will apply the suggested fix, along with the other fixes I detailed in another post for Open Tx 2.1.7. I haven't heard from Jochen in a while now. The project seems to have gone quiet.

athertop, 2016-01-17:
Pull request issued above

athertop, 2016-01-19:
The fix now merged for this issue

closed 2016-01-19 by athertop