Clooney82/MavLink_FrSkySPort #54: Compitability with PX4 flightstack

careyer, 2016-02-19:
Hi there,
since Arducopter development has been slow recently and might come entirely to an end (3DR has terminated contracts with most of the arducopter lead developers) i would love to see this project adapted to the PX4 flightstack. Many of the developers have changed to PX4 now and PX4 is currently getting a huge boost in means of updates, features, useability and Co. Looks like PX4 will be the 1st choice in medium- to long-term.

I have tried the solution on a Pixhawk with PX4 but it does not work. I think this is because currently there is no way to configure the Serial-parameters for the TELEM Ports in PX4.
Probably this code can be adapted to work with PX4 out of the box?

Best regards and many thanks in advance.
- careyer

lvale, 2016-02-19:
Hi
The development on ArduPilot is not slow, on the contrary. It might appear to you, but what is being done will bring severe advantages for the next versions, but it takes time and huge processes of validation and testing. Oh, and the team, although some elements are not being sponsored like before, continues its work as always. Remember that it has always been normal that the ArduPilot team would contribute for the base stack (NuttX /PX4). PX4 has recently gained more "traction" but that doesn't mean that ArduPilot is dead, on the contrary.

If you want to use FrSky telemetry you can use it on PX4 now, like you can now, even without this project on Ardupilot.

Why don't you open an issue on PX4 to suit your requirements? 

Best regards

Luis

closed 2016-02-19 by lvale

rmackay9, 2016-02-24:
I'm one of the main developers of ArduPilot and just to backup a bit what Luis has said above,  Ardupilot is still alive and moving forward.

The project hit a major road bump with 3DR reducing its support but that's not going to kill the project. 3DR continues to support a couple of the developers and other companies have also stepped in over the past year. Intel has well over 10 developers devoted to DroneCode projects including at least 2 working on Ardupilot full time and a few others improving QGroundControl to work better with Ardupilot. Parrot has 1 or 2 developers and there are three smaller companies that have devoted 1 developer each as well. We also continue to have code contributions from independent developers.

So I'm not saying it's all happy days, 3DR has been our major supporter for years but ArduPilot does not rely on 3DR for it's survival. We want ArduPilot to be widely adopted and developed by lots of individuals and companies . This is healthier than being dominated by a single player.