Clooney82/MavLink_FrSkySPort #55: No GPS data and energy calculations shown

lucamax, 2016-03-05:
Hi,
Teensy 3.2
OpenTx 2.1.17
3DR Pixhawk

Looking at Telemetry page (12) on the Taranis , the sensors seems to be there , as i.e. the Gps position, but  they did non show in the display when using  coptr217.lua .
- No Gps fix or data
- No Watt, mAh ,Wh calculations, 
- no flighttime, no max alt
- Heading, speed, altitude  , yes .
Using Telem1.lua generate a memory error

Really do not know how to solve it.
I check again and again but could not find a solution .

athertop, 2016-03-06:
Hi,  on the telemetry sensors page, do you see the different sensor values updating?  Each update sees a * flash next to the changing value. If they are not updating, then you likely have missed out setting the srx_ parameters  in mission planner. The x in srx lines up with the telem port your teensy is connected to. In my case I use telem 2 so amend SR2_ values. Did you first create your telemetry sensors by doing a discovery or manually set them up? Discovery works best, but you will need to rename Temp1 to T1 and Temp2 to T2 as these are named wrong. Discovery is the only way to get a working GPS sensor lat-long format.

lucamax, 2016-03-06:
Hi, thanks for the reply !
yes ,values are updating , I see flashing *  next to each sensor
I use too telemetry  port 2 and SR2 values have been set up in Mission Planner 
yes, I did discovery of sensors after deleting all sensors.
I did not rename Temp1 to T1 and Temp2 to T2
So your suggestion to rename them gave me a huge inspiration .
I manually match the telemetry settings of my taranis with the images of telemetry settings provides in MavLink_FrSkySPort-s-c-l-v-rc-opentx2.1    package .
Some parameters were simply missing , as i.e. SWR,  others had odd settings.
After fighting a bit with the Taranis , both entering right parameters directly in the radio and with Companion because sometimes the Taranis did not accept inputs or had in a certain moment two Curr parameter ;
I was finally able to have everything work :)

My mistake was that I did not check via Companion matching with parameters on BMP in the package.

Thanks again athertop for your help and inspiration .

closed 2016-03-06 by lucamax