Clooney82/MavLink_FrSkySPort #56: Arduino compiling problem

Luftpumpa252, 2016-04-01:
Hello when arduino is compiling telemetry code, arduino error occur.

Arduino: 1.6.7 (Windows 10), TD: 1.27, Vývojová deska: "Teensy 3.2 / 3.1, Serial, 72 MHz optimized, US English"

FrSkySPortTelemetry:116: error: token "Â" is not valid in preprocessor expressions
#if defined USE_SINGLE_CELL_MONITOR ||Â defined USE_FLVSS_FAKE_SENSOR_DATA
^
exit status 1
token "Â" is not valid in preprocessor expressions

athertop, 2016-04-01:
That "Â" character does not exist in the source code. Is your desktop maybe doing some strange language translation perhaps?

Luftpumpa252, 2016-04-01:
Hm, edit #if defined USE_SINGLE_CELL_MONITOR || defined USE_FLVSS_FAKE_SENSOR_DATA to 
#if defined USE_SINGLE_CELL_MONITOR||defined USE_FLVSS_FAKE_SENSOR_DATA and now is working.

athertop, 2016-04-01:
I am using the freshly downloaded source, and it compiles perfectly for me on a Mac using Arduino 1.6.4. It appears you have a newer version of Arduino, but those spaces should nor majke any difference. Your original post stated the error was due to a stray character in the line (FrSkySPortTelemetry:116: error: token "Â" is not valid in preprocessor expressions), but you don't mention this above. Did you initially download the zip file and unpack it to get the source files onto your PC? or did you cut/paste from the source in Github (this could introduce stray chars, so downloading the zip avoids this).

gnkarn, 2016-04-01:
Try this, go to the line with the error indicated by compiler
Delete the line completely, 
Type it again,.

That would resolve that problem ,

Saludos

athertop, 2016-04-05:
The issue is now resolved

closed 2016-04-05 by athertop