Clooney82/MavLink_FrSkySPort #60: telemetry pass through in opentx

dk7xe, 2016-05-03:
there will be a telemetry passthrough functionality in openTX in the future. See statement at https://github.com/ArduPilot/ardupilot/issues/1587#issuecomment-216375018 
and pull request https://github.com/ArduPilot/ardupilot/pull/3434
as well as https://github.com/opentx/opentx/pull/3326 and https://github.com/opentx/opentx/pull/3426 on openTX side.
This is made to overcome the "misuse" of frysky telemetry ID's as it is done for this project also.
I would suggest that you jump on that as well because it does not require a sensor lookup anymore and also sending parameters back from openTX to FC is supported.

lvale, 2016-05-03:
Currently that 3434 is just a PR. There are other developments being made

athertop, 2016-12-22:
For now this is not a priority, and as little documentation available from C&T, this will be readdressed in the future. as this is not an issue per se, I will close. Note that I am keeping an interested eye on the passthru functionality, but I would only consider implementing this personally if the passthru encoding code (that which is currently being handled in Ardupilot 3.4 by setting SERIAL_PROTOCOL=10) is handled by the teensy instead from its regular Mavlink input. This way, we can make the teensy ground based (using tech such as Ultimate LRS, where it passes a full Mavlink stream to the ground over the RF link), making use of both Mavlink (for Mission Planner) and passthru s.port (for the Taranis screen) telemetry.

closed 2016-12-22 by athertop

lvale, 2016-12-22:
From what I understand from the doc that C&T presented the 5000 range of sensor id's is being used, although I believe there's a problem where one can fill all the telemetry slots. https://github.com/ArduPilot/ardupilot/issues/4744

Their doc is waiting to be merged on the ardupilot wiki https://github.com/ArduPilot/ardupilot_wiki/pull/635

I believe that this solution bypasses the Teensy and only requires the inverter.

athertop, 2016-12-22:
Luis, yes, I understand that the current solution merged into Ardupilot no longer requires a teensy (just an inverter). My suggestion above about having a version of this passthru protocol which runs on a teensy stems from two thoughts:

1. I really like the idea of having this passthru S.port telemetry received at the Taranis as it really alleviates the pain which would normally be involved in the Taranis telemetry configuration - making it easier for users to get the solution up and running.

2.  my own involvement in the 'Ultimate LRS' community has allowed me to fully appreciate how useful it is to have ground-based Mavlink (I'm not talking about 3DR radio link, but instead sending the telemetry over the LRS solution to a serial port on the Tx in the Taranis...and at 19.2k baud). By having Mavlink at the ground level, it allows me to hook up a fully functioning Mission Planner connection on the ground (using a BT connection from laptop to the ULRS Tx), whilst at the same time, allowing me to feed that same Mavlink output into a ground-based Teensy (also integrated in the ULRS Tx), which currently runs the sketch provided in this repo to provide the regular LUA screens you see in this repo, to run on the Taranis (simultaneously alongside the aforementioned Mission Planner). So all-in-all a very powerful telemetry ground station setup combining both Mavlink and FrSky S.Port from the same data downlink.

Of course If I simply configured Ardupilot to send down the FrSky passthrough protocol over this serial link, then I would no longer have access to Mavlink on the ground - so this solution doesn't work in my case sadly.

So this lead to my suggestion in my previous post of combining the benefits of both 1 & 2 above to produce a really powerful telemetry solution. The idea would be to have Pixhawk still output regular Mavlink (so not FrSky passthrough directly),and have this downlinked to the ground where it would feed into the Teensy...then have the FrSky passthrough protocol implemented on the Teensy so that it is able to feed the Taranis and allow the simple setup solution as per the C&T solution.

So the key is being able to implement the passthru protocol on a Teensy instead of it happening inside Ardupilot. Does that make sense? Sound plausible even?

lvale, 2016-12-23:
get me a contact so that I can reply privately :)

athertop, 2016-12-23:
Luis, 
I am athertop on rcgroups forum. Feel free to pm me on there. Thanks

gnkarn, 2016-12-27:
Hi , gentleman , i will appreciate some help , here 
After updating my teensy and my taranis to the latest versions of both , my taranis has scrambled some RC channels , and i can't find the error as channel assignment was never changed by me .

one strange symptom that happened upfront , was during telemetry discover , all slots were full ( delete averting first) , so i have to turn off the taranis ,  as it kept looping on that error page .

then, channels  different than mentioned on the wiki, for example Fuel appears four times , not two .

Do you have an actual print page of the Telemetry configuration, instead of what is shown on the Wiki ?
other recommendation to debug ?

thanks

gnkarn, 2016-12-27:
setup : pixhawk 3.4
open tx 2.1.9
and latest version of teensy and lua  files .

athertop, 2016-12-27:
gnkarn, When you say that the 'RC channels' were scrambled - this is a little confusing as the LUA code is produced to only display a telemetry screen - it does not do any channel configuration for your model. I suspect this is a separate unrelated issue. Do you have any other LUA scripts in use within this model configuration (any defined on the Custom Scripts page or other telemetry scripts not provided by this project)?

As for the Fuel sensor, the code is written to only create two fuel sensors on different instance numbers (as per wiki - instances 8 & 9), and the one with instance 9 needs to be renamed  to 'MSG'.  If you are not interested in the second sensor here (this sensor is a tad experimental after all), then you can remove this from the teensy code altogether by commenting out the line #define SEND_STATUS_TEXT_MESSAGE (line 143) in MavLink_FrSkySPort.ino before compiling/writing the teensy sketch. In this case you will not need to add the txtmsg.lua script to Screen 2.

At the sensor discovery stage of the process, as the wiki suggests, its often easier to delete all the sensors first then have them discovered from scratch to eliminate any issues like this - the number of sensors configured is less than 32 so you should never see the slots full message. So the discovery should quickly add all the sensors within a matter of seconds, and then click stop discovery once the list stops growing.

gnkarn, 2016-12-27:
Thankyou for your response,
It is evident that something , is not right  as my lúa code should be the same , I don't understand why I have more slots on telemetry , I willl verify my teensy code  first .

Scrambled means for example that now a knob I was using to change the intensity of my leds , it is now changing the " effect" , it is of course assigned to other channel , and at the same flight modes switches don't respond as intended .

Before updating all was working fine.

Could it be that the taranis new version is not compatible with lúa or teensy code?

Gustavo

athertop, 2016-12-27:
Gustavo, I have recorded a video today showing the telemetry setup. I need 
to still edit before uploading.
I honestly don't know if the non-plus Taranis has less memory than the plus 
version.i will attempt to find out. I only have the plus model myself.

Just to reiterate, the lua code provided with the project does not mess 
with a model configuration aside from configuring a timer. It doesn't touch 
the mixer configuration in any way.

You definitely need to install the latest teensy code for the telemetry 
screen to work properly. You also need really to delete all sensors and 
discover them from scratch as per the wiki instructions on page 1.2.

gnkarn, 2016-12-28:
@athertop ,my taranis is a plus , so this should make the difference, and understand that the lua  code is not the cause, by most probably the teensy , code, with the lua ,  considering i m coming from a version that is a year old, and that i have modified , in order to make my leds on the Y6 to work .

i deleted all sensors , more than once , so my next move is to compare the teensy codes , focusing on  the telemetry areas , and see what can i find , ....

gnkarn, 2016-12-28:
here is the forked code, im working on 
https://github.com/gnkarn/MavLink_FrSkySPort-CLOONEY82-21_LEDv4

athertop, 2016-12-28:
Gustavo, The only affect the teensy code has on the Taranis is to present telemetry sensors, so again, not sure how this would cause your RC channel issues.

I thought I had narrowed down the syntax error messages to a potential code issue tonight (all to do with order of execution and initialisation of variables), so made some changes to remove any possibility of those potential issues. This however made no difference, in fact seems to have made it slightly worse, as even though I have plugged a potential hole, in doing so it meant adding a couple of lines of code which has made the footprint slightly bigger and I am now seeing the syntax error screen more than ever. This strengthens my belief that it is a memory issue. What is rather annoying, is that I never see this issue in the Companion simulator running the same model/script. I only wish the sim properly modelled the Taranis architecture - would make life far easier.

I have been offered assistance to implement some techniques to reduce the memory footprint from a very helpful chap in the advanced lua forum on RCGroups, so hopefully things will improve in the new year.

gnkarn, 2016-12-28:
one strange thing is that i have duplicates IDs after automatic reading telemetry values.
first as mentioned earlier , all 32 channels are full, and in fact would be more, because it saturates ..
id 10 , have one for hug, and other for AccX , and cels , Fuel
, ID 25 have 3  sensors : SWR, RSSI, RxBt
ID5 : AcY , A4,  Fuel, Cels, VFAS, TMp1, A3
ID19: Tmp1,  A3
ID11 :GPS , ACCZ
ID13:Fuel
ID24: VSPD, Cels, GPS, AccX, Aspd,
ID16:fuel
ID8: Galt, AccY,A4, Fuel, 

So , there is no way it can work fine 
i don't know what to do from here, as this so strange ..

gnkarn, 2016-12-28:
should i have //#define POLLING_ENABLED , commented in or out ?

athertop, 2016-12-29:
Should be commented out unless you are using ULRS instead of Frsky X series Tx/Rx. The wiki details this - Section 2 says "For using #define POLLING_ENABLED and #define USE_MAV_RSSI in the code above, see the section "Ultimate LRS Support" for further details." And then the Ultimate LRS page details when you would want to have this enabled. If ou are using a regular X4R/X8R with teensy connected to the s.port then definitely disable the polling option (as per the default config). Its possible having this enabled with X series Rx could cause the duplicates - although I have never tested this configuration (as it is not a valid config).

I fully tested sensor discovery just yesterday by following the Wiki myself to make sure I hadn't missed anything and it worked perfectly - no duplicates - instead 25 discovered sensors ( a couple of these need tweaking after discovery (also detailed in wiki), and then I added the 4 calculated ones manually (also detailed in wiki). I have never seen duplicates created and I have tested every version of teensy code since I joined this project, so wondering if your setup differs in some way?

gnkarn, 2016-12-29:
ok, i found 2 problems, which are "part" of the causes
one, when discovery it is important to select ignore instances ( otherwise duplicates will be generated)
the second problem, is that something changed , since the previous version and this one, on channel assignments , now channels , needs to one less number ( CH-1)
so now things start to make sense

still , problems ,
my taranis do not stop to say "stabilize active" and "invalid flight mode"
without a single pause ...
battle continuous tomorrow, thanks

athertop, 2016-12-29:
Gus, 
"Ignore instances" should not be checked when doing the discovery. Like I said before - some sensors do share the same instance number by design. I have no idea why you are getting more than 32 discovered - this is not the normal behaviour.

I am also confused what you mean by channel assignments. Can you please advise what options you have enabled in the teensy code under "Enable addons/options" ?

athertop, 2016-12-29:
Gus, can I just check - you are running my code (from Clooney82 repo) - as opposed to your own forked version?

gnkarn, 2016-12-29:
I'm running my version, but they are the same , the only difference is the leds code that is my own.

The other problem I see is that the second page , messages is not receiving any ..
Time to sleep, thanks .

gnkarn, 2017-01-03:
everything , seams to be working fine now ,
i think the next step is to make messages to work, i see that the messages development is still based on a variant of the original, by l vale, and others  calaborating , but for some reason there is something different on the actual version that is not working .

i see that all messages are preloaded into one file , instead of looking into it , as it is on Scott´s version  that seams to working very fine .

So the actual "pre-coded" version , should be much easier to code and to make it work  , probably more difficult to maintain .
 so , teensy , just needs to send , one number to taranis , and taranis should pick the message string based on that number , but the thing is i could find where the LUA script, is supposed to load the text messages ..

Could you please elaborate a little , so i can follow from there .

thanks
gustavo

athertop, 2017-01-03:
Hey Gus,
Can you please elaborate on the messages not working?

There are two message systems actually in use:
1. is the voice messages. The teensy does indeed send a numeric code, contained within the Tmp2 sensor. It generates the code numbers from the Avg_statustext_v3.3.ino file. The lua simply plays the msg file corresponding to the number received.
Is this part no longer working?? If not, then maybe there are changes in AP2.4 I need to consider. It worked fine in 3.3.2.

2. is the new txtmsg screen - as the wiki advises, this is still experimental, the teensy encodes the actual message text numerically (letter by letter) and sends it through that way. The LUA script (txtmsg.lua) decodes those messages and translates back to raw text which it displays on the screen. I found that by changing the stream frequency (of the configured streams for the majority of the other Mavlink data) from 10Hz down to 5Hz, this puts less stress on the code and allows the messages to be less corrupted. Its far from perfect though! 
To be honest, I wasn't involved in the development of any of the message code at all (was written by Clooney who is no longer involved in the project), so I don't fully understand how it was coded or the design decisions made. He actually wrote his own c library to do the coding for the txtmsg.lua screen at the Teensy side - way beyond my understanding - I am a relative beginner to all this!

Cheers, Paul

gnkarn, 2017-01-04:
Paul ,  thanks for your support,

i was referring to the text messages , i have not tested the voice messages yet , if they work fine , that could be even better than a text message , so will try and comment later .
in  case they were not good enough i will start studying  the text messages code .
cheers