Clooney82/MavLink_FrSkySPort #62: Findings for improoving the documentation and function

WeareGermans, 2016-05-26:
Hello,

After working with Taranis and Telemetry for the whole day i want to share my findings.
I already edited some pages to transfer some knowledge, i hope that is ok.

Nonetheless there are some findings i can not decide the solutions on:

1. There exist different tables with telemetry values under [0 Overview](https://github.com/Clooney82/MavLink_FrSkySPort/wiki/0.-Overview) and [1.2.1 Telemetry Value Overview](https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.2.1-Telemetry-Value-Overview), that does not support clear documentation. Please delete the older one and update the "recent" one.

2. The pictures under [Taranis Telemetry](https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.2.-FrSky-Taranis-Telemetry) should be updated or the corresponding openTX Version mentioned.

3. There are some undocumented or not correct documented values:

 - ASpd is not documented at all.
 - "VFAS" is written "Vfas" in the puctures and tables, which wont work.
 - GSpd and SWR are "t.b.d." in the table.
 - For me A2 shows HDOP *10 instead of HDOP *25

4. A3 and A4, at least for me, do nothing, they always have the value of 301.
I am using the recent version of Clooney82-MavLink_FrSkySPort with a Pixhawk with 3.3.2 and X8R and Taranis with OpenTX 2.1.8.

So long,
keep up the good work!

athertop, 2016-05-26:
Hi, I am in the process of editing the Wiki to bring it up to date. My efforts so far can only be seen on my own fork of this repository here: https://github.com/athertop/MavLink_FrSkySPort/wiki
If you were willing to proof read my changes I would be very grateful - you obviously have an eye for detail! Once I've finished with the edits and the other devs are happy with the changes, then I will move them over to the main Wiki here to replace the one which currently exists.
Thanks again, Paul (aka athertop)

WeareGermans, 2016-05-26:
Hi Paul,

Great that you do work on the documentation, which is absolutely essential.

I looked a little bit into your fork. Do not take the feedback personal!

Please do chop the first sentence of the overview down to a easier readable one.

Try to use optical separators or listings, nobody likes to read a wall of text. ;-)

Other than that great input!
Pleas do also update the content as to my findings.

Kind Regards,
Franz

athertop, 2016-05-26:
Franz, I totally appreciate your input so don't be worried about any offence :-) I will take your comments on board. I have just been doing some more work on the Wiki pages. My next page to work on is 2.1.2 (the offsets screen written by Wolke)

athertop, 2016-12-05:
A lot of work now completed on Wiki which I believe addesses all reported issues in OP. Will now close.

closed 2016-12-05 by athertop