Clooney82/MavLink_FrSkySPort #71: offset.lua syntax error and wrong flight mode indication

kilwater, 2016-12-08:
Hello, I recently managed to get Your telemetry working on taranis using teensy.  
Unfortunately I get syntax error when trying to edit offset.lua (I'd like to change it for a plane setup).
The other problem is that I get wrong flight mode information on Taranis screen (for ex. - i see Stabilised when I'm on Manual etc.)
Can you help me out ?

Thanks 
Peter

athertop, 2016-12-08:
Peter, all of those issues are fixed in today's release v1.8. see the 
release notes fie that version as there are a number of changes.

Cheers, Paul

kilwater, 2016-12-09:
I'll let you know as soon as i get home after work. Thank you for a fast reply.

kilwater, 2016-12-09:
I believe it's Arduplane v. 3.7.1

kilwater, 2016-12-09:
Ok, soo....
I updated scripts on Taranis and teensy program (with libraries). 
After connecting everything togehter, my Taranis starts to loop audio messages "Invalid flight mode"(male voice) - "Stabilize active"(female voice), on and on...
Changing flight modes doesn't help.
When trying to execute Scripts/Mixer/offset.lua I get Script: Syntax Error. 

What I've tried so far:  Updated newest firmware on Taranis, ereased SD Card and copy files from Your package on it - without success. 

Peter

athertop, 2016-12-09:
Peter,
OK, I know that this script does work with Plane 3.7.1 as that's what I 
have. Mine runs on pixhawk rather than APM. Unfortunately, I don't have an 
APM to test it on, but I don't think this is the cause of your issue.
The files supplied with the project are additional files to the standard 
Opentx 2.1 files, so you need those files also.

Can you also check the configuration of your telemetry sensors. I have seen 
similar flight mode issues before fixed by deleting the Fuel sensor and 
rediscovering it. Make sure you have sensor 8 configured as 'Fuel'  and 
sensor 9 named MSG , and not mix these sensors up.

As for the Syntax error when running offset.lua, can I just check that you 
added that script to the lua scripts page of your model, and that you are 
not executing it any other way? I do occasionally see syntax errors display 
on the main telemetry screen but that is due to a memory issue in opentx, 
and to get round this its just a case of repowering the Taranis to reload 
scripts.

I personally run these on a Taranis Plus, and not the earlier non-plus 
model Taranis. They do run also in the Taranis simulator though.

Let me know how you get on. Cheers, Paul

kilwater, 2016-12-09:
I'll let you know in 20 minutes if I'll manage to fix it. Thanks

kilwater, 2016-12-09:
Alright - I've uploaded new CD card files.
There are no looped voice messages anymore.
As for sensors - for some reason there is no sensor named MSG - instead there is another Fuel sensor with id 9 (values jumping from 0 to 32768%)
First Fuel with id 8 is present.

I uploaded a video with this issue:
https://youtu.be/bHi3rGhEc1M

Taranis is set to polish language but I believe it's not a problem. 
Flight modes are very different form what is really set.

P.s. sorry for my English.

athertop, 2016-12-10:
Peter,
If I can reiterate my previous comment please: "Make sure you have sensor 8 configured as 'Fuel'  and sensor 9 named MSG , and not mix these sensors up."

This is also detailed in the wiki here: https://github.com/Clooney82/MavLink_FrSkySPort/wiki/1.2.-FrSky-Taranis-Telemetry
(search for MSG on that page and you will find the details on how to configure the MSG sensor).

The issue arrises as there are two Fuel sensors (sensor ID 0600) defined by the Teensy code - one uses sensor instance 8 - (we use this for flight mode) and another using sensor instance 9 (we use this for status messages). Unfortunately OpenTx just sees these both as sensors called 'Fuel' - so the discovery adds both with the same name - the second one (with instance 9) needs to be renamed to MSG so it can be unique from the first.

Hope this helps. I will also try to update the wiki to make it clearer about the sound files and combining these with the OpenTx 2.1 sound pack.

Cheers, Paul

closed 2016-12-22 by athertop