Clooney82/MavLink_FrSkySPort #76: CELLS CONFIGURATION Zx

gnkarn, 2016-12-30:
i suggest to include more detail on the Cells sensor configuration . for Z1 to Zn where n is number of cells.

type :calculated
formula :Cell
Cell sensor : Cels
Cell index :  1 to 6 ( put the same number as the Zx

athertop, 2016-12-30:
Hi - I mention in the wiki that this is an optional configuration as it is not used by the telemetry script at all. Taking this in regard I think there is enough info to be honest.

closed 2016-12-30 by athertop

gnkarn, 2016-12-30:
i don't get it , im using the cells info, and is part of the project as an "add-in", so why not to explain how to configure it properly ?

i have almost all working , now ( on the discovery process i confirmed, that at least in case , the option for ignore instances "HAS to be checked" .

now only things not working fine, is Text messages , and feedback of mode change. tks

athertop, 2016-12-30:
Sorry, I presume you mean the single cell monitor using a resistor network connected directly to the teensy analog ports? If that's the case then I am not aware that the z1..zn sensors are mandatory for that to work.  Instead I think that all is required is a Cmin calculated sensor which will reflect through to the main telemetry page, keeping a monitor of the minimum cell voltage in the flight pack (or 'packs' if you have 2).

I tried that addin once in the early days of this project (before I got involved in a dev capacity) but quickly moved to using a FrSky FLVSS sensor instead as I found it far more practical. I wasn't involved in the development work of the single cell monitor (I believe it was Michael Wolkstein who did that bit a long time before I got involved in the project).As a result, I have had little exposure to that part of the code.

The 'ignore instances' option may have to be checked for your modified code to work, but it certainly shouldn't be checked for the code in THIS repo for the sensors to be discovered properly. I have no idea why your code is behaving differently from the main repo code if it is truly the same code - I suspect you must have some differences in there given the requirement for the ignore instances option.

Mode change messages also work fine on the current repo lua code. They have on all versions to be honest, so again, not sure why they don't work in your version. The text is displayed on the main telemetry screen by the toppanel() function, and the voice is called by the Flight_modes() function which is run as part of the 'background' loop. So two different functions manage these two aspects.
Its important to note that the code caters for flight modes from both the Plane and Copter AP firmware versions. If you are running the offset.lua mixer script then you can use the Coptr/Plan: option to choose - this sets a global variable called gAPType, where a value of 1 refers to Copter and 2 refers to plane. If you do not have this mixer script installed, then the main.lua script takes care of assigning the default for this global variable to 1 (the checkGlobals() function does this). This value is important as it is used as the index to which sub-table is referenced in the FlightMode table for flight mode names. It also decrees which wav file suffix is applied for when the flight modes are called out. You really need to check your code to make sure all the required bits are in place to allow this to work. It certainly works fine in the published code in this repo. ( I have been using it in the simulator all day).

The text on the txtmsg screen is still very experimental and often results in corruption. The reason for this is that these messages can not be read in time if the LUA script is busy doing something else. For this reason I have changed the teensy code and the wiki to detail that Mavlink stream rates should be reduced to 5hz instead of 10hz. I think I may just remove this txtmsg screen from the repo until a better way is developed for accessing these messages.

gnkarn, 2016-12-30:
the , single cell monitor in my setup is working perfect, so no issues here,
i will take a closer look at messages, as i saw other projects with seams to be working fine , may pick some idea .. , thanks

athertop, 2016-12-30:
Gus, I would suggest, just as a test that you install just the Clooney82 repo files rather than your own copy and see that they work OK before moving on to your own versions.
I set these up from scratch yesterday on a new model (hexacopter) and it worked perfectly - It was my attempt to follow the Wiki to make sure it was correct, and I can confirm that it worked perfectly. I have nothing special in my setup - I am using the default teensy code with no addons enabled and the FLVSS LiPo sensor from FrSky. Standard Pixhawk using Arducopter 3.4.3, OpenTx 2.1.9.