Clooney82/MavLink_FrSkySPort #78: Request or need help to setup

Dutchy-79, 2017-02-14:
Hi,

On a video from Peter King (from droneshop) I noticed he had a last known position screen, So if the quad/plane gets lost you van type those GPS cords in Google Maps and go try and find it.

Is this already in this version? If so pls help me setting it up if not, I'd like this function to be added.

Regards,

Jasper

athertop, 2017-02-14:
You are maybe referring to the GPS coordinates from the model telemetry screen,rather than our lua telemetry screen. You can also check the log checkbox for the GPS sensor setup to record the GPS coordinates to the SD card logs. This would allow you to see the last known coordinates.

closed 2017-02-14 by athertop

Dutchy-79, 2017-02-14:
Thanks for closing this so quick,.. 

I meant this video @ 5:25 https://www.youtube.com/shared?ci=QABaNvmxoSo

athertop, 2017-02-15:
Last known GPS position in that video is being shown on one of the standard Opentx 2.0 telemetry screens. On OpenTx 2.1 you need to create the telemetry screen layouts yourself, but just add the GPS telemetry sensor as one of the values and you have a similar layout to what Peter was showing in that video.

Dutchy-79, 2017-02-15:
Thanks, got it.
And is it possible to get a voice notification if the quad comes back to home for some reason?

VFAS I can setup but the fence or failsafe i don't know how.

athertop, 2017-02-15:
If the quad comes home, then it will have changed to Return To Launch mode, so the voice will notify "Return to Launch Active".
There is nothing in the code for Fence specifically. There is a mechanism for verbal status messages to be called, but it only covers messages upto Ardupilot 3.3 at present. I am working on a re-write of that entire area of code so that it includes all messages for all AP versions newer than 3.2. It may take some time though. All the new development is going on in my own repo just now, not this one (which just has the legacy code now). My repo is here: https://github.com/athertop/MavLink_FrSkySPort

Dutchy-79, 2017-02-15:
On the telemetry screen id does change but the voice isn't activated, I used all the sound files in the original master brance, I'll try you fork.

athertop, 2017-02-15:
By default the teensy code is  for ardupilot 3.3 or newer and Opentx 2.1. You will need the correct sound files installed. The repo in my new fork is reorganized so please check my new version wiki for details. Especially in relation to the fuel sensor renaming.

athertop, 2017-02-15:
Dutchy-79, you need to be sure that the fuel sensor on ID 9 is renamed to MSG or this will interfere with the flight modes - the wiki explains this here: https://github.com/athertop/MavLink_FrSkySPort/wiki/1.2.-FrSky-Taranis-Telemetry

Dutchy-79, 2017-02-16:
Thank you for the heads up, I did setup MSG 

I'm using a apm 2.5 with opentx 2.2RC10 so maybe that's the problem. Gonna try your fork when back home

athertop, 2017-02-16:
Ah, so with APM, you are using Arducopter 3.2? In that case you will need the 3.2 sound files and also make some tweeks in the ino file before writing the teensy sketch. 

reverse the comments on these two lines in MavLink_FrSkySPort.ino:
//#define AC_VERSION          3.2      //line 128
#define AC_VERSION          3.3       //line 129

Then see the wiki section about choosing the right version of sound files for AC3.2

Dutchy-79, 2017-02-16:
Yup INFO MissionPlanner.MAVLinkInterface - 16-2-2017 10:09:25 1 **ArduCopter V3.2.1** (36b405fb)

Just got home, uploading now.

athertop, 2017-02-16:
In that case you definitely need to be using the old messages so the changes I suggested earlier. I am working on consolidating all the message code/files to have a one-fits-all approach in a future version.

Dutchy-79, 2017-02-19:
One issue left I guess, the Wh and the vertical  battery bar don't reset on reset flight nor reset telemetry. 

First time I went out with second battery and not turning off the transmitter, just used reset flight instead. 

Then you have to fly with a beep beep beep transmitter lol

athertop, 2017-02-19:
Thanks for testing this - I will need to reproduce it to figure out why this is different on OpenTx 2.2. 
Can I please ask first though: Did you restart your Taranis between flights? And if you did, did it retain the Wh value even after the restart? I suspect not.

These consumption values are simply held in memory as the script runs, and so only reloading the script will zero these - the only way to reload the script s to restart the Taranis. This was the same in OpenTx 2.1. The script remains in memory for the duration even when you are not on the telemetry screen. Its a good idea to restart the radio in any case between flights. Also after changing model its a good idea to restart the Taranis as this ensures any LUA script is not retained in memory from the previous model.

I am just in the process of reorganising the sound files - this is just the prep for the major consolidation exercise, but as part of this I have released my own versions of the OpenTx Sound Packs, using both 'Ava' and 'Samantha' (Apple Mac) voices. If you are using APM with AP3.21 then the Ava voice is the best one for you as the Telemetry sounds for APM3.2 are provided in this same Ava voice. I will publish these soundpack changes to the release branch shortly with updated instructions in the Wiki. Presently the sound file changes are only in the -dev branch.

Dutchy-79, 2017-02-19:
Hi, Thanks.

No I did not turn off the TX, Normally I use long press enter and then reset flight, I also tried to reset telemetry. I will power cycle the Taranis next time. Just thought it might be a bug, Glad it's not.

Looking forward to your soundpack,

athertop, 2017-02-19:
Initial Soundpack changes now implemented in the master repo - and also documented in the wiki here: https://github.com/athertop/MavLink_FrSkySPort/wiki/1.1.-SD-Card-Contents

Dutchy-79, 2017-02-20:
Uploading them now, Thanks