Clooney82/MavLink_FrSkySPort #79: Script syntax error

MattMgn, 2017-10-11:
Hello,

First, thanks for sharing all this work. 
I am trying to use MavLink_FrSkySPort for my taranis, I am using OpenTx 2.2 and Ardurover on a Pixhawk 1.
I have followed wiki and I can detect sensors, and edit them as mentionned. However, when I run main.lua script, I get a `Script syntax error` (the screen is displayed for 1 sec before getting error message). 
I am quite new to lua script and opentx in general, any help for debugging would be helpfull.

Matt

athertop, 2017-10-11:
Hi Matt, 

Firstly, the script has not been developed to run for the Rover AP firmware - it contains flight modes for Plane and Copter only. It could run with some modification for sure, but out of the box it will report the wrong flight modes running it on Rover. There are also no wav files for the Rover flight modes in the project either sadly.

Secondly, to get to your issue - Script Syntax Error is likely being caused by a memory issue on the Taranis trying to run this version of the code. The version of the project found here in the Clooney82 repo is really only for OpenTx v 2.1 (not 2.2). I know the Wiki mentions 2.2 rc8, but the script has since been further developed to run on the release version of 2.2 in a new repo which you can find here: https://github.com/athertop/MavLink_FrSkySPort. The version in the new repo has been broken up into modules, and compiled - all done to reduce the memory requirement, and it runs fine on OpenTx 2.2 release. So feel free to take a look at the new Repo - it has a much more up to date Wiki also on the new repo, which will hopefully help. Cheers, Paul A (athertop)

closed 2017-10-11 by athertop

MattMgn, 2017-10-14:
Hi athertop,

Thank you for this answer. I would want to use this script for Ardurover, thus I am wondering what kind of modifications are needed. You mentionned FLIGHT MODES, that mean it could be possible to modify current FLIGHT MODE names to match those used by Arducopter, right ?
I presume, each flight mode has its own wav files, which is read when changing FM. I could generate missing wav files for ardurover. But I am really not confident about other modifications.  I have identified some of them  : 
        - fmodes.lib : add apType == "Rover" 
        - SOUND/en/TELEM : add AVFM{flight_mode}R.wav files

Could you tell me if you see other modifications. 
Also, I am about to test your fork of Mavlink_FrskySPort, on my current OpenTx 2.2 version. Thank again for support.

Matt

MattMgn, 2017-10-14:
I also see modification to main.lua
```
    if shvars.apType == 0 then
	    WavSfx = "A"
	  elseif shvars.apType == 1 then
	    WavSfx = "P"
	  else
	    WavSfx = "R"
	  end
```