Clooney82/MavLink_FrSkySPort #81: Precision/Threshold

stratus35fr, 2017-11-18:
Hi,
I detected a little bug with  2.1.9 version of the X9D  taranis.
In order to control the max power allowed to the engine, I created a logic switch that compares "Power" to a fixed value of 1400W (Engine MAX Power). I use a FLVS and Current Sensor devices to calculate the Power.
If I change the precision of the calculated variable "Power" in the telemetry window (0. or 0.0 or 0.00) the value of 1400W is assigned a ratio 10. The threshold was thus raised to 14000W !!

athertop, 2017-11-19:
This repo is unsupported now - please feel free to check the repo here: https://github.com/athertop/MavLink_FrSkySPort
Not entirely sure how I can help with your issue though - its not related to my LUA code,  ut to LUA scripting with OpenTx - so you maybe want to raise an issue with OpenTx or on the RCG forum for LUA.

closed 2017-11-19 by athertop

athertop, 2017-11-19:
I just remembered that if you change the sensor precision, then you need to change the LUA code to compensate. For example, you will see in this project Cmin has a precision of 2 so 0.00, and as a result the LUA code has to include PREC2 in the line:
lcd.drawNumber(61,10,getValue("Cmin")*100,MIDSIZE+PREC2), as well as being multiplied by 100. I think its a workaround to treat all telemetry values low level as integers.